FreeCAD/FreeCAD #16224: The 2D Draft button on the Start page ignores the defaultWP preference

Roy-043, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The "2D Draft" button on the Start page ignores the defaultWP preference. It calls Std_ViewTop even if defaultWP is 2 (Front) or 3 (Side).

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-07-19:
Still an issue.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/ea33a15aaf97909d5ad51507c8b7c4587c73f93f

The 2D Draft button reads defaultWP and picks the matching front or side view instead of always using top.