FreeCAD/FreeCAD #16227: Using 0.22.0dev Union of 2 solids fails

cropduster32, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating the Archimedes Fan (by mangojelly) the final fusion of the two solids failed with the error msg "Resulting solid is invalid".
The following is shown in the report view:
20:07:35  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/snap/freecad/1123/usr/Mod/Part/BOPTools/BOPFeatures.py", line 88, in make_multi_fuse
    self.copy_visual_attributes(obj, obj.Shapes[0])
  File "/snap/freecad/1123/usr/Mod/Part/BOPTools/BOPFeatures.py", line 106, in copy_visual_attributes
    target.ViewObject.ShapeAppearance = source.ViewObject.ShapeAppearance
<class 'AttributeError'>: 'Gui.ViewProviderLink' object has no attribute 'ShapeAppearance'
20:07:41  Fusion: Resulting shape is invalid

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38577 (Git) Snap 1123
Build type: Release
Branch: main
Hash: 2a9cf9e8e6e62086948eb15e24d1e69a54f53d97
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-02:
Can you share the file?

luzpaz, 2024-09-02:
An aside, looks like the Snap isn't using latest coin 4.0.2

cropduster32, 2024-09-13:
[Archimedes-Fan (FCStd).zip](https://github.com/user-attachments/files/16998982/Archimedes-Fan.FCStd.zip)

Is this any good? Just change the zip to FCStd