FreeCAD/FreeCAD #16228: SIGSEGV crash when trying to recover a file

tomaszn, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashed when I was editing a file.
Now a crash occurs when I reopen FreeCAD and try to recover my file.
The stack trace is attached below.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.5.28
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Anything else?

21:36:12 systemd[2826]: Started app-gnome-freecad-1084608.scope - Application launched by gnome-shell.
21:36:14 gnome-shell[1084617]: /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
21:36:14 gnome-shell[5019]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x360000e specified for 0x3600019.
21:36:27 gnome-shell[1084617]: Reading failed from embedded file: StringHasher.Table.txt (3916 bytes, 674 bytes compressed)
21:36:28 gnome-shell[1084617]: Reading failed from embedded file: PartShape14.bin (0 bytes, 2 bytes compressed)
21:36:28 gnome-shell[1084617]: Reading failed from embedded file: PartShape16.bin (0 bytes, 2 bytes compressed)
21:36:28 gnome-shell[1084617]: Reading failed from embedded file: PartShape18.bin (0 bytes, 2 bytes compressed)
21:36:28 gnome-shell[1084617]: Reading failed from embedded file: PartShape20.bin (0 bytes, 2 bytes compressed)
21:36:28 gnome-shell[1084617]: Reading failed from embedded file: PartShape44.bin (0 bytes, 2 bytes compressed)
21:36:29 gnome-shell[1084617]: Reading failed from embedded file: PartShape47.bin (0 bytes, 2 bytes compressed)
21:36:29 gnome-shell[1084617]: Reading failed from embedded file: PartShape48.bin (0 bytes, 2 bytes compressed)
21:36:29 gnome-shell[1084617]: Reading failed from embedded file: PartShape52.bin (0 bytes, 2 bytes compressed)
21:36:29 gnome-shell[1084617]: Reading failed from embedded file: PartShape53.bin (0 bytes, 2 bytes compressed)
21:36:29 gnome-shell[1084617]: Program received signal SIGSEGV, Segmentation fault.
21:36:29 gnome-shell[1084617]: #0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x77c787e45320]
21:36:29 gnome-shell[1084617]: #1  0x77c78a27af50 in Data::ElementMap::dehashElementName(Data::MappedName const&) const from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x90
21:36:29 gnome-shell[1084617]: #2  0x77c78a27cd5b in Data::ElementMap::getElementHistory(Data::MappedName const&, long, Data::MappedName*, std::vector<Data::MappedName, std::allocator<Data::MappedName> >*) const from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x30b
21:36:29 gnome-shell[1084617]: #3  /tmp/.mount_freecatFfbi1/usr/lib/./Part.so(+0x2a7839) [0x77c7512a7839]
21:36:29 gnome-shell[1084617]: #4  0x77c7512a81bb in Part::Feature::getRelatedElements(App::DocumentObject*, char const*, Part::HistoryTraceType, bool) from /tmp/.mount_freecatFfbi1/usr/lib/./Part.so+0x1bb
21:36:29 gnome-shell[1084617]: #5  0x77c751426c50 in Attacher::AttachEngine::calculateAttachedPlacement(Base::Placement const&, bool*) from /tmp/.mount_freecatFfbi1/usr/lib/./Part.so+0x100
21:36:29 gnome-shell[1084617]: #6  0x77c7512bc2ed in Part::AttachExtension::isAttacherActive() const from /tmp/.mount_freecatFfbi1/usr/lib/./Part.so+0x8d
21:36:29 gnome-shell[1084617]: #7  0x77c7512be67e in Part::AttachExtension::extensionOnChanged(App::Property const*) from /tmp/.mount_freecatFfbi1/usr/lib/./Part.so+0xde
21:36:29 gnome-shell[1084617]: #8  0x77c78a062219 in App::ExtensionContainer::onChanged(App::Property const*) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x59
21:36:29 gnome-shell[1084617]: #9  0x77c78a06078c in App::DocumentObject::onChanged(App::Property const*) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x11c
21:36:29 gnome-shell[1084617]: #10  0x77c78a17a835 in App::Property::hasSetValue() from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x135
21:36:29 gnome-shell[1084617]: #11  0x77c78a1c198e in App::PropertyLinkSubList::setValues(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&&) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x12e
21:36:29 gnome-shell[1084617]: #12  0x77c78a1c2075 in App::PropertyLinkSubList::Paste(App::Property const&) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0xe5
21:36:29 gnome-shell[1084617]: #13  0x77c7512be7de in Part::AttachExtension::onExtendedDocumentRestored() from /tmp/.mount_freecatFfbi1/usr/lib/./Part.so+0x2e
21:36:29 gnome-shell[1084617]: #14  0x77c78a05b2b4 in App::DocumentObject::onDocumentRestored() from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0xd4
21:36:29 gnome-shell[1084617]: #15  0x77c7444b4ec3 in Sketcher::SketchObject::onDocumentRestored() from /tmp/.mount_freecatFfbi1/usr/lib/Sketcher.so+0x13
21:36:29 gnome-shell[1084617]: #16  0x77c78a041de1 in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x811
21:36:29 gnome-shell[1084617]: #17  0x77c78a042e7d in App::Document::afterRestore(bool) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0xbd
21:36:29 gnome-shell[1084617]: #18  0x77c78a2363b1 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADApp.so+0x1c81
21:36:29 gnome-shell[1084617]: #19  0x77c78b273e20 in Gui::Dialog::DocumentRecovery::accept() from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADGui.so+0x720
21:36:29 gnome-shell[1084617]: #20  /tmp/.mount_freecatFfbi1/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x77c7884eb964]
21:36:29 gnome-shell[1084617]: #21  /tmp/.mount_freecatFfbi1/usr/bin/../lib/libQt5Widgets.so.5(+0x2fab48) [0x77c7894fab48]
21:36:29 gnome-shell[1084617]: #22  /tmp/.mount_freecatFfbi1/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x77c7884eb964]
21:36:29 gnome-shell[1084617]: #23  0x77c78945b522 in QAbstractButton::clicked(bool) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libQt5Widgets.so.5+0x32
[... 24 lines cut ...]
21:36:29 gnome-shell[1084617]: #48  0x77c78b27718e in Gui::Dialog::DocumentRecoveryFinder::checkForPreviousCrashes() from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADGui.so+0xae
21:36:29 gnome-shell[1084617]: #49  0x77c78b090e02 in Gui::Application::checkForPreviousCrashes() from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADGui.so+0x32
21:36:29 gnome-shell[1084617]: #50  0x77c78b5dcbba in Gui::MainWindow::delayedStartup() from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libFreeCADGui.so+0x47a
21:36:29 gnome-shell[1084617]: #51  0x77c7884e0830 in QObject::event(QEvent*) from /tmp/.mount_freecatFfbi1/usr/bin/../lib/libQt5Core.so.5+0x260
[... cut ...]
21:36:29 gnome-shell[1084617]: #64  /tmp/.mount_freecatFfbi1/usr/bin/freecad(+0x7620) [0x5789eb270620]
21:36:29 gnome-shell[1084617]: #65  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x77c787e2a1ca]
21:36:29 gnome-shell[1084617]: #66  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x77c787e2a28b]
21:36:29 gnome-shell[1084617]: #67  /tmp/.mount_freecatFfbi1/usr/bin/freecad(+0x7981) [0x5789eb270981]
21:36:29 systemd[1]: tmp-.mount_freecatFfbi1.mount: Deactivated successfully.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amreo, 2024-09-02:
Where is your file?

benj5378, 2024-09-02:
I'm not sure if corrupted files can be considered a bug. We can fix what causes FreeCAD to freeze, but we can't really fix corrupted files. To get your file back, open the FCBak file.

tomaszn, 2024-09-02:
The file can't be shared.

Thank you for your suggestion about the backup file.

luzpaz, 2024-09-02:
Note: you aren't using the latest development revision of 0.22.x

closed 2024-09-02 by luzpaz