FreeCAD/FreeCAD #16229: Mouse click on body does not attach reference in Change the Linked Object

Forresto0, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TNP? When I am editing a previous pad operation, the pad face changed. I was following a video tutorial (link below). After clearing the old reference in the Link Dialog Window, I was unable to add a reference with the mouse.

The steps for referencing the new face was to 
>>click sketch001
>>Property View
>>Attachments
>>Support
>>...
>>Link Window
>>Select the pad in question
>>Clear
>>Mouse Click the desired face

No reference was made through the mouse click. 
Editing the Map Mode did result in the desired Pad Face to be referenced.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.15
  * QuickMeasure 2022.10.28
  * SearchBar 1.0.2
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

YouTube Tutorial showing assumed correct FreeCAD behavior:
https://www.youtube.com/watch?v=B7DOIxgwAAk
Navigate to 4:50 in video, if desired.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-02:
@Forresto0 please test on dev version and report back https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

 We aren't backporting fixes to 0.21.x as we are close to releasing 1.0

Forresto0, 2024-09-02:
@luzpaz Sorry, didn't know you weren't looking for 0.21.x issues for windows. 1.0? I thought 0.22 was next release.
Tested on dev version and it WORKS! (Version: FreeCAD_weekly-builds-38555-conda-Windows-x86_64-py311)
Reference updated with mouse clicks, no problem. 

Text overlap still exists in Property View>>Attachment Support field. See pic.
[image]

luzpaz, 2024-09-03:
@Forresto0 no worries. Thank you for reporting. Hopefully FreeCAD release cycles get shorter after the 1.0 release (which will come after 0.22.x)

> Text overlap still exists in Property View>>Attachment Support field. See pic.

@maxwxyz is there a ticket for this ?

closed 2024-09-03 by luzpaz