FreeCAD/FreeCAD #16232: Fresh (2.9.) regression: Crash in Gui::LinkView::linkGetElementPicked

azuk, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- Load file bus - Copy-tn.FCStd in forum post https://forum.freecad.org/viewtopic.php?p=778287#p778287
- Hover mouse over the model in 3D view
- Observe that FreeCAD crashes

Alternatively the problem can be reproduced with BIMExample.FCStd that comes with FreeCAD, but you may need to do a bit more hovering.

I bisected the problem to commit 1822312127865f73ee7ae0d48b6c4b9e27a33b9c by @PaddleStroke 

Backtrace (info->linkInfo is null on line 1397 in ViewProviderLink.cpp):

```
#6  0x00000791d823b507 in __assert_fail (assertion=0x791de223c35 "px != 0", 
    file=0x791de1cc120 "/usr/include/boost/smart_ptr/intrusive_ptr.hpp", line=201, 
    function=0x791de2116e0 "T* boost::intrusive_ptr<T>::operator->() const [with T = Gui::LinkInfo]")
    at ./assert/assert.c:103
#7  0x00000791ddf8690e in boost::intrusive_ptr<Gui::LinkInfo>::operator-> (this=0x5c22c7783da8)
    at /usr/include/boost/smart_ptr/intrusive_ptr.hpp:201
#8  0x00000791ddf69a48 in Gui::LinkView::linkGetElementPicked (this=0x5c22cc5db500, pp=0x5c22cb3df060, subname="")
    at /removed/freecad/src/Gui/ViewProviderLink.cpp:1397
#9  0x00000791ddf7124f in Gui::ViewProviderLink::getElementPicked (this=0x5c22cb4baee0, pp=0x5c22cb3df060, 
    subname="") at /removed/freecad/src/Gui/ViewProviderLink.cpp:2310
#10 0x00000791ddf850d2 in Gui::ViewProviderFeaturePythonT<Gui::ViewProviderLink>::getElementPicked (
    this=0x5c22cb4baee0, pp=0x5c22cb3df060, subname="")
    at /removed/freecad/src/Gui/ViewProviderFeaturePython.h:257
#11 0x00000791ddd1391d in Gui::SoFCUnifiedSelection::getPickedList (this=0x5c22cc085550, action=0x5c22cca505f0, 
    singlePick=true) at /removed/freecad/src/Gui/SoFCUnifiedSelection.cpp:269
#12 0x00000791ddd17322 in Gui::SoFCUnifiedSelection::handleEvent (this=0x5c22cc085550, action=0x5c22cca505f0)
    at /removed/freecad/src/Gui/SoFCUnifiedSelection.cpp:747
#13 0x00000791d7a7e818 in SoAction::traverse(SoNode*) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#14 0x00000791d7c38628 in SoChildList::traverse(SoAction*, int, int) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#15 0x00000791d7c8a8a1 in SoGroup::doAction(SoAction*) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#16 0x00000791d7ca9dbc in SoSeparator::doAction(SoAction*) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#17 0x00000791d7a7e818 in SoAction::traverse(SoNode*) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#18 0x00000791d7a8c15b in SoHandleEventAction::beginTraversal(SoNode*) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#19 0x00000791d7a7ea0c in SoAction::apply(SoNode*) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#20 0x00000791d7c56028 in SoEventManager::actuallyProcessEvent(SoEvent const*) ()
   from /lib/x86_64-linux-gnu/libCoin.so.80c
#21 0x00000791dde1b756 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent (this=0x5c22cca64460, 
    event=0x5c22cc7cc220) at /removed/freecad/src/Gui/Quarter/QuarterWidget.cpp:1009
#22 0x00000791dde25784 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent (this=0x5c22cca64460, 
    event=0x5c22cc7cc220) at /removed/freecad/src/Gui/Quarter/SoQTQuarterAdaptor.cpp:752
#23 0x00000791dde7eebf in Gui::View3DInventorViewer::processSoEventBase (this=0x5c22cca64460, ev=0x5c22cc7cc220)
    at /removed/freecad/src/Gui/View3DInventorViewer.cpp:2576
#24 0x00000791dde3c7ef in Gui::NavigationStyle::processSoEvent (this=0x5c22ca683070, ev=0x5c22cc7cc220)
    at /removed/freecad/src/Gui/NavigationStyle.cpp:1522
#25 0x00000791dde42f8f in Gui::CADNavigationStyle::processSoEvent (this=0x5c22ca683070, ev=0x5c22cc7cc220)
    at /removed/freecad/src/Gui/CADNavigationStyle.cpp:325
#26 0x00000791dde3c6dc in Gui::NavigationStyle::processEvent (this=0x5c22ca683070, ev=0x5c22cc7cc220)
    at /removed/freecad/src/Gui/NavigationStyle.cpp:1493
#27 0x00000791dde7ee7d in Gui::View3DInventorViewer::processSoEvent (this=0x5c22cca64460, ev=0x5c22cc7cc220)
    at /removed/freecad/src/Gui/View3DInventorViewer.cpp:2571
#28 0x00000791dde134f6 in SIM::Coin3D::Quarter::EventFilter::eventFilter (this=0x5c22cad53a40, obj=0x5c22cca64460, 
    qevent=0x7fff4d2ce460) at /removed/freecad/src/Gui/Quarter/EventFilter.cpp:167
#29 0x00000791d8cd7e82 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00000791d996bd35 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#31 0x00000791d9973ecb in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00000791dd8b52e0 in Gui::GUIApplication::notify (this=0x7fff4d2cf720, receiver=0x5c22cb9b5800, 
    event=0x7fff4d2ce980) at /removed/freecad/src/Gui/GuiApplication.cpp:83
#33 0x00000791d8cd8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00000791d9972874 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#35 0x00000791d99caa39 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#36 0x00000791d99cdfbf in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#37 0x00000791d996bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#38 0x00000791dd8b52e0 in Gui::GUIApplication::notify (this=0x7fff4d2cf720, receiver=0x5c22c91781a0, 
    event=0x7fff4d2cee90) at /removed/freecad/src/Gui/GuiApplication.cpp:83
#39 0x00000791d8cd8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#40 0x00000791d9145a3b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#41 0x00000791d9117bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#42 0x00000791d29c3d06 in ?? () from /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#43 0x00000791d6b145b5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#44 0x00000791d6b73717 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#45 0x00000791d6b13a53 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#46 0x00000791d8d35279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#47 0x00000791d8cd6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#48 0x00000791d8cdf3e8 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#49 0x00000791dd707657 in (anonymous namespace)::tryRunEventLoop (mainApp=...)
    at /removed/freecad/src/Gui/Application.cpp:2165
#50 0x00000791dd7079b0 in (anonymous namespace)::runEventLoop (mainApp=...)
    at /removed/freecad/src/Gui/Application.cpp:2185
#51 0x00000791dd707d8e in Gui::Application::runApplication ()
    at /removed/freecad/src/Gui/Application.cpp:2262
#52 0x00005c22c5c89cae in main (argc=1, argv=0x7fff4d2cfe68)
    at /removed/freecad/src/Main/MainGui.cpp:301
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Debug
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1.dev
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * Render 2024.8.4
  * Curves 0.6.44
  * Assembly4 0.50.15
  * Assembly3 0.12.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-02:
Thanks for the detailed ticket. 

>  Fresh (2.9.) regression: Crash in Gui::LinkView::linkGetElementPicked

Question: What do you mean by 2.9. ?

azuk, 2024-09-02:
> Question: What do you mean by 2.9. ?

Date when the problem appeared (when the commit was merged). I'll remove that from the title to avoid confusion, not important.

> > Backtrace (info->linkInfo is null on line 1397 in ViewProviderLink.cpp):
> 
> Is this the line ?
> 
> https://github.com/FreeCAD/FreeCAD/blob/941656bcd3f82f147c9b8e1d8840f8c463858b06/src/Gui/ViewProviderLink.cpp#L1396-L1397

Yes.

PaddleStroke, 2024-09-03:
It was 
```
        if(nodeIdx == it->second)
            ss << it->second << '.';
        else
            ss << info.linkInfo->getLinkedName() << '.';
```
And in order to fix https://github.com/FreeCAD/FreeCAD/issues/15822 I changed it to : 
```
        ss << info.linkInfo->getLinkedName() << '.';
```
In my testing I didn't get a crash and it was working correctly. Now I guess there are some conditions where info.linkinfo is null apparantly.

PaddleStroke, 2024-09-03:
So the crash is caused by the Darft array object. It happens when it is a 'Link array'.

Simplified file to reproduce crash.

[bus - Copy-tn.zip](https://github.com/user-attachments/files/16845101/bus.-.Copy-tn.zip)

PaddleStroke, 2024-09-03:
So it happens that draft link array is using LinkElements (if you change the property "Expand Array" you see them in the tree).

Then we have 2 solutions :
- Option 1 : We exclude the draft array from the fix by doing : 
```
        if (!info.linkInfo)
            ss << it->second << '.';
        else
            ss << info.linkInfo->getLinkedName() << '.';
```

This way draft array go back to the situation they had before. (ie https://github.com/FreeCAD/FreeCAD/issues/15822 is not fixed for them)

- Option 2 : Draft Arrays are fixed so that link.infoLink is not null anymore.

I will start by making a PR for option1 as it will prevent this crash. If @yorikvanhavre wants to have a look in Option2 it will still be possible.

luzpaz, 2024-09-03:
//cc @Roy-043 JFYI

Jajcus, 2024-09-08:
I have similar crash in my project (too complex to share). Looking forward for the fix being merged (not tested it myself, though).

closed 2024-09-08 by chennes