FreeCAD/FreeCAD #16233: transparency impropperly rendered when using save image / Std ViewScreenShot

Cinnabar48, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when using "save image", transparency is created where a transparent object sits in front of a non transparent object.
Sections of high brightness are also punched out and saved as transparency.

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) AppImage
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * woodworking 0.21.2.33771
  * freecad.gears 1.2.0
  * Estimate 0.1.2
  * A2plus 0.4.65
  * fasteners 0.5.25
  * Assembly4 0.50.15
  * Curves 0.6.42
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?p=768157
[image]
[image]
[transparency test FSCtd file.zip](https://github.com/user-attachments/files/16841400/transparency.test.FSCtd.file.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Cinnabar48, 2024-12-03:
just updating to say this issue still persists (as of FreeCAD_weekly-builds-39416-conda-Linux-x86_64-py311.AppImage). my guess would be that the cause of the overlay problem is when the image is created; instead of layering each element, the properties of each object starting at the back are applied to the whole image and this causes transparency to be applied to whole sections of the image even if there are solid objects behind it.

tarman3, 2024-12-03:
Confirm
[image]

<details>
  <summary>1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

FlachyJoe, 2024-12-03:
ref: https://github.com/FreeCAD/FreeCAD/blob/77ba6eed4f1b007b9b0d0883e7432866e5fee381/src/Gui/View3DInventorViewer.cpp#L2126-L2195

@wwmayer you wrote this 6 years ago, do you know how to improve it?

FEA-eng, 2026-11-02:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```