FreeCAD/FreeCAD #16234: Balloon annotations misplaced on rotated views

johnsonm, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@WandererFan — In #15726 I reported, as part of what seemed a similar problem, that balloon annotations did not point correctly in rotated views. This part my my initial report was not fixed in #16075, so it clearly wasn't the same problem. Repeating it here separately to clarify:

Rename [Cap.zip](https://github.com/user-attachments/files/16464073/Cap.zip) to Cap.FCStd, then open the drawing. Try to attach an annotation to the selected point in the middle of the left hand side of this section view:

[image]

The arrow of the resulting annotation points elsewhere:

[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Unknown
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-09-03:
Am I correct in only seeing this on initial placement of the balloon?  Dragging the whole balloon (arrow and bubble) seems to put the arrow tip in the correct place.

johnsonm, 2024-09-03:
Correct — I do confirm that #16075 fixed dragging the whole thing afterward. It's only the initial placement for rotated views (at least this one!) that I'm reporting here as not fixed from what I previously reported in #15726.

Thanks!

commit 4e655a88 (2024-09-09): [TD]fix initial balloon placement on rotated view (fix #16234)
commit 54eb7b3d (2024-09-09): [TD]fix initial balloon placement on rotated view (fix #16234)
closed 2024-09-09 by yorikvanhavre