FreeCAD/FreeCAD #16235: Sketcher: CreateArcSlot bug with direct measurement enty

pixelwaster, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue has been present in the last 3 dev version. It is not present in the ,21 version because arc slot doesn't exist.

1 select arc slot tool
2 click once to set the center point for the arc
3 type 30 and hit tab/enter for a 30 mm arc
4 type 100 and hit tab/enter for a 100 degree arc length

Here is where the bug starts. The next entry box should be for the slot width (flat ends) or radius (arc ends). Instead the box contains the negative sign and the arc length. -100 degree if 100 was entered and -230 if -100 is entered.
If the mouse is left clicked at this point the arc slot with be in two parts and over contained,
But if you move the mouse just a tiny amount left along a horizontal direction, the bottom part of the arc slot with jump from the center to the end of the arc slot. If you move the the right by the same amount on the horizontal, the end will jump back to the center. 
Move the mouse to the left again, then left click. You will be able to enter the width/radius and tab/enter for a complete arc slot.
The move left is subtracted from the main radius and this is the number in the text box AFTER the left click. Moving along a horizontal will make the flat/arc width smaller until is passes though zero and gets larger. Left click now will set the width of the end/arc and the arc slot will finish. You can also enter a number and tab/enter to finish the arc slot.

tab/enter after Step 4
[image]

slight move to the left (notice the lower part of the slot has moved)
[image]

left click
[image]

enter the width flat/radius
[image]

moving the tool all the way to the left after clicking
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * CAM
  * Draft
  * Fem
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Klevente12, 2024-09-02:
try this
https://bit.ly/4e4zLlq
select the desired compiler in the installer

pixelwaster, 2024-09-02:
Antivirus warning.

luzpaz, 2024-09-06:
@pixelwaster thanks so much for the detailed report. Is there any chance you could make a quick screencast of the issue so devs can quickly grok what you're reporting ?

pixelwaster, 2024-09-06:
I'll ty to get a 'cast posted. Need to install some software.

pixelwaster, 2024-09-06:
https://github.com/user-attachments/assets/7bedb054-3457-429a-9004-2f9701839085

pixelwaster, 2025-10-04:
The tool is working just fine. The status bar messages are a nice help.

`OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/112.198/1
Installed mods: 
  * AddonManager 2025.9.23
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * FreeCAD-themes 2025.6.4
  * freecad.gears 1.3.0
  * IconThemes
  * ImportNURBS 1.1.0
  * lattice2 1.0.0
  * Manipulator 1.6.3
  * OpenTheme 2025.5.20
  * pcb 6.2023.1
  * Plot 2024.11.26
  * Quetzal 1.5.6
  * retr3d
  * sheetmetal 0.7.55
  * Silk 0.2.0
  * sketchershapes
  * Telemetry 1.0.2
  * Vars 0.0.2.beta5
  * Woods 1.0.2
`

closed 2025-12-02 by pixelwaster