FreeCAD/FreeCAD #16236: PartDesign Boolean: Does not work all the time.

leoheck, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This operations has some issues.
- Sometimes the operation does not work, regarding the order of the objects selected.
- Sometimes the refine does work.

See the attached example here: [part_boolean_fail.zip](https://github.com/user-attachments/files/16842907/part_boolean_fail.zip)

`Test2` fuses 2 cylinders. The operation works, but the result is not right. 
However (`Test3`) if I invert the order of the fused objects the operation results in an expected object.

Both `Test2` (right) and `Test3` (left) should have the same height.

[image]

The Body `Bottle_Neck` shows an operation that does not `refine`. 

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-03:
Does it work in the stable 0.21.2 version? What is the refine setting by default?
@bgbsww FYI

AIRCAP, 2024-10-08:
the booleans do exactly what they are supposed to do

the issue in the example file - if there is one - is where the sketches respectively pad objects end up. the "non working" example has the cylinders perfectly overlapping, so that the fusion of both is equal to both of the originals, while the third example has them offset, creating a larger fusion

this is not a bug in boolean, at least not in the latest version

maxwxyz, 2024-10-09:
done

closed 2024-10-09 by maxwxyz

leoheck, 2024-10-09:
> suggestion, close this bug unless a file for reproduction which has an actual boolean malfunction is provided

@AIRCAP The file you are looking for is in the first post.