FreeCAD/FreeCAD #16237: Opened macro path shows on tab in some circumstances

samohd, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening a macro via the Execute Macro > Edit button shows filename without path on the tab. This is good. Opening by drag/drop into FC or via the Start tab shows the full path. Uses too much space and is unnecessary.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-03:
Can you please add a screenshot of what you're referring to ? (Tip: Drag/drop the image into github textbox for auto-attach to ticket)

samohd, 2024-09-03:
[image]

samohd, 2024-09-03:
Probably don't need "- Editor" on the tab either, since FCMacro gives that away.

paulcobbaut, 2024-09-14:
Same issue on Debian with Python files.

When opening "one.py" the name of the tab includes the (very very long) full path to the file.

```
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.62
  * GIS-main
  * Trails 2022.1.0
  * Curves 0.6.22
  * LCInterlocking
  * CurvedShapes 1.0.5
  * fasteners 0.5.0
  * ToolbarStyle
  * Manipulator 1.5.7
```

luzpaz, 2024-09-14:
> Opening a macro via the Execute Macro > Edit button shows filename without path on the tab. This is good. Opening by drag/drop into FC or via the Start tab shows the full path. Uses too much space and is unnecessary.

What's the proposal here ?
* Make sure drag/drop and Start tab options work the same as opening via Execute Macro 
* Remove  "- Editor" from the tab since FCMacro implies it

Is that correct ?

paulcobbaut, 2024-09-14:
Never showing the path would be perfect... for me.
Maybe the best option is a check mark in Preferences-General-Document that says 'Enable/Disable path in tab-name'.

samohd, 2024-09-14:
The typical way to do this is show the filename on the tab, and the full path on a hover popover display.

luzpaz, 2024-09-15:
@Roy-043 quick question: what's the procedural code to make a dynamic tooltip (that shows a file path) ?

Roy-043, 2024-09-16:
> what's the procedural code to make a dynamic tooltip

I don't know.

paulcobbaut, 2025-07-19:
Downloaded the latest build AppImage and the issue is still the same. I looked through the preferences but find no setting to disable showing the path.

luzpaz, 2025-09-19:
@mwganson any chance you'd be interested in doing this ?

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/69aa97cf4ca7454926d160dd0c364f0bff717d55

Files opened from the Start page or by drag and drop use the filename-only tab title too.