FreeCAD/FreeCAD #16238: Transform tool use deactivates assembly

tibfel, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I use the translate tool it deactivates the assembly, so I am at the root of the document. This happens with some other commands as well, but I cannot remember them from the top of my head. I often find myself clicking in vain somewhere, only to realize that I am out of the workbench at the root of the document. 

Could there be an optional feature that would give the user a message that the current workbench is to be deactivated with yes/no options? When you create a sketch and sketcher is activated, there should be no message, but many times I get out of Techdraw or Assembly to the root document for no valid reason and this is undesired behaviour. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-03:
I can see that Std_TransformManip deactivates an assembly, but I do not see a workbench switch.

Related: #15819.

tibfel, 2024-09-03:
I did not phrase it correctly. Out of the workbench meant that tools that should be accessible in the relevant workbench are no longer accessible, e.g. I cannot drag unconstrained/underconstrained parts in assembly, because I am out of the assembly. I need to double click assembly to activate it and then I can drag parts again. This happens many times in various scenarios ( I have fixed some of them by unchecking pressing esc can leave the edit mode in the preferences). 

For me, the best fix would be if FreeCad would introduce different file formats for Part, Assembly, Drawing, etc. Once I am in a drawing file, only techdraw commands will be available and I cannot get out of it. If I want to edit a part from a drawing, I will right-click and open the document and it will open in the part environment.

I saw in the past, that the idea of introducing separate file formats for drawings, assemblies and parts was dismissed and fCStd will be used as a master container for everything.

PaddleStroke, 2024-10-08:
Currently, using the transform tool deactivate the assembly on purpose.
The reason is that the transform tool requires the 'edit'. So the assembly cannot keep the edit.

IMO the transform tool should not require the edit. I don't know if it is technically possible or not.
Also the transform tool should be merged with the placement tool (see https://forum.freecad.org/viewtopic.php?t=90891). Perhaps that would be a good time to tackle this issue as well.

closed 2024-11-29 by WandererFan