FreeCAD/FreeCAD #16239: PartDesign: Pocket problem

Pendelf, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Pocket command does not work if you snap to an axis (point on object) when creating a slot
When you delete Point on object and create it again, everything works fine
https://github.com/user-attachments/assets/045b5d86-b090-4587-a603-ca3b24668ff5

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.28
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.8.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-03:
Confirmed. I think this is primarily a sketcher issue. Opening the original sketch and wiggling some geometry also fixes the problem. Seems the same issue as #16198.

FEA-eng, 2026-04-25:
I can't reproduce it. @Pendelf Can you recheck?

[image]

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260421 (Git shallow)
Build date: 2026/04/21 23:21:45
Build type: Release
Branch: (HEAD detached at 449725202)
Hash: 449725202d0ab581516aacad59d62cdce80db82a
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```