FreeCAD/FreeCAD #16241: Draft: Snap Intersection snaps to endpoints

semhustej, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Snap Intersection is turned on, the tool snaps to endpoints of lines/polylines too. This creates confusion and some problematic behavior, when the tool snaps to a point that user is not expecting.

From my understanding when only Snap Intersection is turned on, only intersections should be snapped to. Not other snapping points.

Demonstration of the behavior in the video:
https://youtu.be/_ynIWdD8GiE

Related discussion:
https://forum.freecad.org/viewtopic.php?t=90191

Similar behavior in FC 0.21.2 and 0.22

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-03:
I also see, that the same setting snaps to corners of rectangles/polygons/closed polylines. I am not sure, whether this is bug too or desired behavior. I would not consider theses points as intersections myself.

luzpaz, 2025-10-16:
@semhustej can you reproduce on latest dev ?

semhustej, 2025-10-16:
Still relevant.
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/07 22:49:55
Build type: Release
Branch: main
Hash: 68f083a6d169ec523fa9d46b7cc40addb1e4ffbe
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
  * UpdatedPythonPackages

```