FreeCAD/FreeCAD #16242: BIM: Error when rotating a column

gianlucafiore, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can rotate this piece counterclockwise, but clockwise through the z axis. I'm using the normal rotate command (R,O).

[image]

testing file
[edificio nuevo.zip](https://github.com/user-attachments/files/16846549/edificio.nuevo.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-03:
In your screenshot we can see an error happening, and in that error, the Draft.rotate command got a wrong parameter (`nan` instead of the angle value). Would that be the cause of the problem? How did you make that happen? I haven't tested your file specifically, but rotating a column seems to work normally for me here, clockwise or not.

gianlucafiore, 2024-09-03:
I do this by using shift to snap to an axis, then rotate by rotating the mouse, and snap again to the perpendicular axis. This clockwise action gives an error

gianlucafiore, 2024-09-03:
I just realized that the column had moved one level down, and when I clicked on a vertex of the base as the starting point of the rotation, and then wanted to snap the rotation to orthogonal angles, it generated a strange snap to the vertices above, and apparently introduced a zero value to the rotation, which then threw the error that the value was incorrect.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

yorikvanhavre, 2024-09-04:
I cannot reproduce this problem on my computer... You'd need to give me precise step-by-step instructions..

But I wonder if the problem is not that you are snapping to points that are at different Z coordinates? Maybe try either to position the working plane appropriately before rotating (and enable working plane snap), or make sure all the snapped points lie on a same plane.

gianlucafiore, 2024-09-16:
> But I wonder if the problem is not that you are snapping to points that are at different Z coordinates? Maybe try either to position the working plane appropriately before rotating (and enable working plane snap), or make sure all the snapped points lie on a same plane.

[image]

By performing the operations you mention, everything is done correctly. I have not been able to repeat the error repeatedly, but I believe that in said screenshot you can read the problem of the entered value, even though it was not the one detailed in the parameter entry window. I understand that it is not the correct way to do it, but the possibility is given, so any user without moderate knowledge could fall into said error.

yorikvanhavre, 2024-09-23:
If I'm understanding correctly, you are picking 1) a rotation pivot that is not on the WP, 2) points to define the rotation angle that are on the WP. That might indeed give pretty incorrect rotation values.

We could fix this by checking for a NaN value and cancelling the operation of course, but it would be better to know the exact cause...

luzpaz, 2025-05-10:
@semhustej any chance you can reproduce this ?

semhustej, 2025-05-11:
> [@semhustej](https://github.com/semhustej) any chance you can reproduce this ?

I am able to reproduce in FC 1.0.

I am not able to reproduce in current windows 1.1dev version, so in my opinion this issue can be closed.

Steps to reproduce:

- select the column in question
- start `Draft_Rotate`
- pick one of the bottom column points as a base point for rotation
- pick the top point of the column which is exactly above the bottom point as a reference angle
- pick the same point as a new angle
- the error message shows up

See the behavior in the video (both. FC 1.0 and 1.1dev)

https://github.com/user-attachments/assets/89437eb6-118a-4c29-ad4c-afdabbcd5c40

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1

```

closed 2025-07-07 by Roy-043