FreeCAD/FreeCAD #16243: Sketcher [request]: Improved trim tool capability (non-destructive operation)

kaiwas, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently we can simply break the line using the "trim edge" tool.
I think it would be useful to have an option that would preserve geometry by automatically creating auxiliary geometry (on the deleted section) and assigning parallel constraints.

We can do this now, but all these actions have to be done manually:
1) Create auxiliary geometry.
2) Select segments
3) Assign constraints.

This process can probably be automated.

Possible concept in the picture. Maybe something else can be added.

[image]

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-11-12:
A simpler version of this would be to have a colinear(tangent) constraint create when a line is split in 2 by a trim.

AjinkyaDahale, 2025-03-13:
I expect the restricted curves I propose [here](https://github.com/FreeCAD/FPA-grant-proposals/issues/33) will provide an alternative.