FreeCAD/FreeCAD #16244: Sketcher: Constraints of equality of circle segments are not assigned.

kaiwas, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I encountered the fact that it is impossible to assign equality to circle segments.
Video demonstration.

https://github.com/user-attachments/assets/803f1ccf-154e-4caf-9103-a88136a424d6

I can't check in version 0.21.

If such a possibility never existed, it could be implemented. Such a method would allow "making a cast" of the length from a circle or spline and assigning it to a straight line segment.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-03:
This is one example of the Solver needing some left field thinking, attached is a file created in 0.21.1, all you need to do is change the diameter (65mm) and the segment length will update using the expression. If you need to change the number of segments then it's awkward/annoying but doable.

Just rename the file from zip to FCStd.
[FourCircleSegments.zip](https://github.com/user-attachments/files/16847843/FourCircleSegments.zip)

kaiwas, 2024-09-03:
> This is one example of the Solver needing some left field thinking, attached is a file created in 0.21.1, all you need to do is change the diameter (65mm) and the segment length will update using [...]

there are cases when you need to assign equality of lengths to two different types of elements.
for example, make a line of such length that it would be equal to a circle.
Yes, I know that this can be implemented through expressions. Or come up with some other workaround (as in many, many places this problem is solved). But all this can be done much more conveniently than propping everything up with eternal crutches.

https://github.com/user-attachments/assets/01a070fe-a0c1-4ac4-a9ca-663374818860

A good user means trigonometry and geometric dependencies, but we want to be lazy, so that the program does basic things on its own. Isn't that right? )

kaiwas, 2024-09-03:
That being said, I appreciate that there are "workarounds" and I'm not calling for them to be removed. In some cases, they are even more useful than simple solutions.

Roy-043, 2024-09-03:
Equality for curved segments refers to their curvature (for circles and arcs this means they have the same radius), not their length.

kaiwas, 2024-09-03:
> Equality for curved segments refers to their curvature not their length. For circles and arcs this means they have the same radius.

I understand that. There is a need to "add" new algorithms. We can get (calculate) the same length of segments with different radii. I don't see any obstacles to doing it in the opposite direction.

Roy-043, 2024-09-03:
OK. Can you update the issue title and text to reflect that?

kaiwas, 2024-09-03:
> OK. Can you update the issue title and text to reflect that?

I did this, but I'm not sure that everything is correct. Can you check? I hope the essence is clear.