FreeCAD/FreeCAD #16245: Add Code Aster for beam analysis.

gsg001, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As discussed with [NewJoker](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=36858) on the [Forum](https://forum.freecad.org/viewtopic.php?t=90182) I am opening an issue to work on the above.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of Ondsel: 64-bit
Version: 2024.2.2.37240 (Git)
Build type: Release
Branch: (HEAD detached at 2024.2.2)
Hash: fbb794cd2e56fba5fd911c8832332b5e5e235319
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * AddonManager
  * Arch
  * Assembly
  * CAM
  * Draft
  * Fem
  * FEM_FrontISTR 0.1.0
  * Help 1.0.3
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * Ondsel-FirstRun
  * Ondsel-Lens 2024.7.5.02
  * OpenSCAD
  * Part
  * PartDesign
  * Plot 2022.4.17
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-03:
@gsg001 Do you happen to what is newjoker's github account name ?

FEA-eng, 2024-09-03:
Any form of support for code_aster in general would be great but beams can be a good start due to the issues related to them (and shells) in CalculiX where they are internally expanded to solid elements.

luzpaz, 2024-09-03:
Who's spearheading this ?

FEA-eng, 2024-09-03:
> Who's spearheading this ?

Hard to say. For now, we have 2 new contributors willing to work on this. I will support them in whatever way I can. I don't have much experience with Code_Aster but I can help with the FreeCAD FEM side.

Perhaps one of the volunteers could apply for an FPA grant (I don't need it personally).

hhassey, 2024-09-05:
I have some coding experience in python, and I do have experience in Code-Aster.  I will help in any way I can.

luzpaz, 2024-09-06:
@FEA-eng any suggestion where @hhassey could begin ?

FEA-eng, 2024-09-16:
> @FEA-eng any suggestion where @hhassey could begin ?

There is a wiki page that can be a good start: https://wiki.freecad.org/Extend_FEM_Module

I would look closely at the implementations of other solvers. If you want to start from beam elements, check solvers that use them (CalculiX, Mystran or Z88). [Mystran](https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/Fem/femsolver/mystran) can be a good start because its implementation is rather basic and includes beams. At the very least the following tools must be supported already:
- mesh (Gmsh/Netgen)
- FEM MaterialSolid
- FEM ElementGeometry1D
- FEM ConstraintFixed
- FEM ConstraintForce

Of course, there will also have to be a solver object for Code_Aster interacting with it.

FEA-eng, 2024-09-25:
https://forum.freecad.org/viewtopic.php?t=90915

trilbytim, 2024-09-25:
Hello! When I checked on the forum a month ago nothing seemed to have happened on Code Aster integration since about 2012, now suddenly there seems to be several of us! I need to use shells (for my use case expanding into 3D elements is not appropriate) and found Mystran a bit limited so turned to CA. I'm also aiming to make a CA reader/writer for FreeCAD (initially aiming at shell elements). I created this fork for working in, although haven't yet got far. https://github.com/trilbytim/FreeCAD  (see forum post above)

trilbytim, 2024-10-17:
Some. I've got a Code Aster Solver object created, although it's not actually doing much useful at the moment.
https://github.com/trilbytim/FreeCAD

luzpaz, 2024-10-17:
@trilbytim Thank you.  
Can you share what you're having issues with ? What you tried and what you haven't ?

trilbytim, 2024-10-17:
@luzpaz My current bugbear is trying to add equations to the Code Aster solver. Although I only intend to be implementing linear elasticity for the time being, given the capabilities of CA it would make sense to have the option to choose between modelling different physical phenomena. This is done with "Equations" for Elmer. I've tried to add an elasticity equation for Code Aster if you see src/Mod/Fem/ObjectsFem.py , you can see I've attempted to get it to add an Elasticity equation object to the CA solver object in the same way as works for Elmer. I've also included the equation files in the codeaster solver folder and modified the CMake lists etc. It's not working though and I've run out of ideas on that for the time being and am just hard-writing linear elasticity into the .comm file for the moment.

Then the big thing I need to work out is how to set the constraints as groups in the mesh. Apparently this is done in Elmer, so I need to look at that. Then get Gmsh to write the mesh as .med

Finally, need to read the results .rmed file and get that back into FC.

So still lots to do, any help gratefully received. Also note, someone who's working with me has got Code Aster to build with a single command now. See the forum thread: https://forum.freecad.org/viewtopic.php?p=787596#p787596

luzpaz, 2024-10-17:
@trilbytim Thanks for the outline! And congrats on the success of building 'codedisaster' (:laughing:) on nix! Don't forget to look at the work these other FreeCAD FEM solver projects have utilized:  
* https://github.com/jaheyns/CfdOF
* https://gitlab.com/FrontISTR-Commons/FEM_FrontISTR

Krande, 2024-10-18:
Hey guys, great to find more people looking into Code Aster!

If it can help in any way we have set up a conda-forge distribution of Code Aster (with all dependencies) which is following the weekly releases by EDF. 

See 

https://github.com/conda-forge/code-aster-feedstock
and 
https://anaconda.org/conda-forge/code-aster

As of today, we only support linux in the conda-forge distribution, but I have also made set up a conda-compilation for Code Aster for windows using MSVC (for reference see https://github.com/conda-forge/code-aster-feedstock/issues/65). 

The MSVC Code Aster conda packages (and all dependencies) can be found at

https://prefix.dev/channels/code-aster

And for what it's worth I have some experimental code aster support in ada-py -> https://github.com/Krande/adapy/tree/main/src/ada/fem/formats/code_aster

luzpaz, 2025-07-29:
Any progress here?

trilbytim, 2025-07-29:
@luzpaz I've transferred my fork into my work's Github account here: 
https://github.com/AMRC-Composites/FreeCAD-CodeAster
The project that I was doing that needed this is now finished, I'm hoping to get more funding for developing it further, but for now all I'm doing with it is trying to keep it reasonably synced up with the main FC repo, I'm not adding any more CA features at the moment. I have not implemented beam elements as this isn't something I use, but I have made it work for shells, including layered shells (composite laminates). You can see how I've done this here: https://github.com/AMRC-Composites/FreeCAD-CodeAster/blob/main/src/Mod/Fem/femsolver/codeaster/add_femelement_geometry.py around line 38 is where you'd have to insert something relating to beam elements.