FreeCAD/FreeCAD #16248: Feature: Draft_Fillet does not remember the last used settings

semhustej, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Priority: low

With most Draft tools, when they are started repeatedly, the last used setting is saved. Draft_Fillet does not save settings when invoking the tool repeatedly. It always defaults to the original settings.
This leads to slower workflow, when user wants to use the same settings on the next fillet, which is quite common.

Suggested change:

- Save Draft_Fillet settings when the tool is started repeatedly. 
- It would be useful, if last used fillet radius was saved too and could be confirmed only by hitting enter. It often occurs in AEC and CAD applications, that the same radius/chamfer is used repeatedly on multiple places.

Settings not saved:
[image]

Video describing the behavior:

https://github.com/user-attachments/assets/2e75e5bb-8e63-4924-8aaf-06b271531382

I could reproduce this behavior on 0.22dev and 0.21.2 version.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-03:
/cc @Roy-043

kaiwas, 2024-09-03:
Let me comment on this.
Saving previously accepted settings is missing in many places (almost everywhere).
In the vast majority of cases, all settings are "zeroed out" and have to be re-entered, even if a repeat action is required. This is very sad.

The zero value of the move (after the move action has already been applied) looks especially funny. Nobody ever moves "to zero" and this is useful only for the initial input. 
And there are many such examples.

https://github.com/user-attachments/assets/0893de0e-8c93-4ef9-95c4-56b9a448d680

semhustej, 2024-09-03:
@kaiwas I mainly reported the first part of the feature suggestion, which would be similar to how other tools work and lead to consistency of all tools, which is desirable. 
I am not sure if remembering values is doable and how hard it would be to implement. I did not check other tools. It would be nice for sure.

kaiwas, 2024-09-03:
> I mainly reported the first part of the feature suggestion, which would be similar to how other tools work and lead to consistency of all tools, which is desirable.

yes. I understand.
It's just that this problem occurs very often. (You will encounter it many more times).
Even if the "continue" checkbox is checked. I don't know how difficult it would be to solve this, but the fact remains. We have to re-enter the data over and over again (even if it is repeated, and this is often required).
There are almost no options to repeat the last action performed.

Roy-043, 2025-01-14:
@kaiwas What you are asking is currently not feasible because the X, Y and Z inputs are updated to reflect the current cursor position. Entering the same values would probably make more sense for a multi copy operation I think.

PR #19067 (2025-01-16): Draft: Save fillet settings
Fixes #16248.
commit 288f3aad (2025-01-16): Draft: Save fillet settings

Fixes #16248.
closed 2025-01-16 by Roy-043