FreeCAD/FreeCAD #16252: Assembly reports multiple AttributeError: getCameraNode errors

furgo16, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open [attached file](https://github.com/user-attachments/files/16847937/Barn.door.zip)
2. The notification area and report panel are filled with these errors:

```python stacktrace
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 1172, in camera_callback_rotation
    self.set_lock_rotation()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 1177, in set_lock_rotation
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 1168, in camera_callback
    factor = self.get_lock_factor()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 1184, in get_lock_factor
    return get_camera_height(self.gui_doc) / 300
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode
Traceback (most recent call last):
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 818, in camera_callback
    scaleF = self.get_JCS_size()
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 889, in get_JCS_size
    return get_camera_height(self.gui_doc) / 20
  File "/snap/freecad/1124/usr/Mod/Assembly/JointObject.py", line 153, in get_camera_height
    camera = activeView.getCameraNode()
AttributeError: getCameraNode

```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38624 (Git) Snap 1124
Build type: Release
Branch: main
Hash: 941656bcd3f82f147c9b8e1d8840f8c463858b06
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Anything else?

There are a few other warnings in the report panel, which go away if the file is saved:

```
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Body
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Sketch
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Body001
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Sketch001
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Body002
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Sketch002
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Pad002
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Pad002
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Pad
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Pad
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Pad001
<App> PropertyTopoShape.cpp(394): Pending recompute for generating element map: Barn_door#Pad001
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-03:
Confirmed. When the file is opened the active view is a TechDraw page and not a 3D view. The code does not take that into account.

furgo16, 2024-09-05:
> When the file is opened the active view is a TechDraw page

Thanks. Unrelated to this issue: I noticed that a while ago and I've always wondered why. What's the reasoning behind showing the TechDraw page first, whenever opening a document? This seems to slow down opening the documents, but it might be my perception only.

Roy-043, 2024-09-06:
I don't know the reason for this. But you can close the tab of the TD page before saving.

furgo16, 2024-09-10:
> you can close the tab of the TD page before saving.

Thanks! I hadn't realized you could do this.

Another issue that this file has, which I don't knof if related, is that whenever it is open, it is marked as modified.

furgo16, 2024-10-06:
Should this issue be added to the 1.0 milestone?

closed 2024-10-14 by chennes