FreeCAD/FreeCAD #16254: Sketcher: Merging vertices by distance

kaiwas, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

At the moment, there is no way to mass merge close vertices in the sketch editing mode. We are forced to manually select pairs using single clicks.

But if there was a tool that would allow you to merge the selection using a frame or all vertices at once within a given radius, it would be convenient.

[image]

An example of work from blender. This often helps out.

https://github.com/user-attachments/assets/e08573c4-34ea-4c6b-a015-1a26f0b083f3

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-03:
The [ValidateSketch](https://wiki.freecad.org/Sketcher_ValidateSketch) tool allows you to find and automatically fix (what you are requesting here) missing coincidences within a specified distance tolerance. But having it available while editing a sketch would indeed be nice.

kaiwas, 2024-09-03:
> The [ValidateSketch](https://wiki.freecad.org/Sketcher_ValidateSketch) tool allows you to find and automatically fix (what you are requesting here) missing coincidences within a specified distance tolerance. But having it available while editing a sketch would indeed be nice.

Yes, I know about this possibility. Thanks. But it would be interesting to move this function to another level.
Also, you can add "only selected" to this.

kaiwas, 2025-09-30:
@JustTryingToGetSomeWorkDone  look this https://github.com/FreeCAD/FreeCAD/issues/24335

JustTryingToGetSomeWorkDone, 2025-10-01:
I'll see if I can find the file and get the problem to reappear, but I can't make any promises. I don't know why it happened the first time.