FreeCAD/FreeCAD #16255: Failing regression test: test_issue_15735

chennes, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current main as of 2 Sept 2024 is failing on the following test:
```
def test_issue_15735(self):
        """
        15735: Point in sketch as loft profile won't work in dev, but works in stable
        The following test is a simplified version of the issue, but the outcome is the same
        """
```
The PR that seems most related is https://github.com/FreeCAD/FreeCAD/pull/15777 -- @PaddleStroke do you have a suggestion for how to proceed here?

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38480 (Git)
Build type: Release
Branch: main
Hash: 20549cacd2d61aec16b6b7873fb73962a6241d3b
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-03:
I guess we should revert #15777 ?

PaddleStroke, 2024-09-03:
This is very easy to fix and was to be expected.
Historically adding a point was forcing it to be construction. Making it impossible to create a normal point. The piont of 15777 was to make it possible.

So this test is doing : 

```
        PointSketch.addGeometry(Part.Point(Base.Vector(0.000000, 0.000000, 0)))
        PointSketch.toggleConstruction(0)
```

Which is adding a point, then toggling construction.
Before it was adding a point as construction, then it was toggled to become normal. Now because it is not specified in the constructor, the point is created as normal geometry. So the toggle is now bad and just needs to be removed.

closed 2024-09-03 by chennes