FreeCAD/FreeCAD #16257: PartDesign: Pad generates a surface if crosses plane.

GAZ082, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For this particular model, if i run a Pad and it crosses the plane, a "surface" is generated.

https://github.com/user-attachments/assets/6503f871-11a8-4426-a0b5-78702d45105e

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.8.17
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Test.FCStd.zip](https://github.com/user-attachments/files/16851253/Test.FCStd.zip)

remove .zip

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2024-09-03:
Looks like this is just a visual issue, i created a mesh based on this body and the surface is not there (or i guess, the resulting thing has a thickness of zero?)
[image]

GAZ082, 2024-09-03:
Looks like if I change the radius of the D to other than 3mm, the artifact dissapears.

https://github.com/user-attachments/assets/135244b4-3e60-4fec-9e04-7e7166592205

luzpaz, 2025-08-14:
@ElementW thanks for your comments and feedback. Would it be possible to please add your full [About](https://wiki.freecad.org/About) info to help us pinpoint what 3rd party deps and other parameters you're running when you tested. TIA!