FreeCAD/FreeCAD #16258: Assembly Joint making can be much simpler with this simple change

samohd, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Assembly is more complex to use than it needs to be in the procedure to add joints. For example to add a simple Slider... It requires too much hunting around to find faces/edges that have the blue arrow pointing in the direction you want the movement. And the inevitably, the end result is completely wrongly offset, so then that has to be manually entered one axis at a time in the Data panel. In this image I managed to find the only one inside edge for the cover, that matches the box, and that resulted in a very easy slider. It began in the exact position I wanted, without having to offset afterward. But not such sensible edge/face exists for the tray. So here you see the blue arrow points the wrong direction for the slide I want. So I have to choose two edges/surfaces that make no sense with each other, and then tweak the Offset afterward. It's also not intuitive that you have to know that only the blue arrow direction matters.

[image]

The simple fix to this is to make it so that you can first choose edges/faces that make sense and will provide immediately correct positioning by the joint. For example in this image, if I choose the bottom face of the tray, and the top face of the bottom of the box - makes complete sense. These will be automatically positioned by the joint, perfectly. BUT... the resultant slide direction will be wrong, because the blue mini axis arrows don't point the right way. Simply making it so that you can change the axis of movement for each object in the joint (to override the default it magically comes up with), would make this all super easy to use.

[image]

There are some more simplification things that can be done. But this is the most critical one to ease of use.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-03:
/cc @PaddleStroke

PaddleStroke, 2024-09-04:
Sorry but I do not understand clearly what you want to do.
You want to be able to select the 2 faces and have a slider joint that slides the tray in and out?
If you can make a screen recording with explanation it would be easier to follow

samohd, 2024-09-04:
No worries, I'll explain more. In the following image, I started a Create Slider Joint. I have selected the underside of the Tray and I have highlighted the upper side face of the bottom of the box (as shown in previous image with two green faces). I didn't click this second face, because that would joint the two objects and you wouldn't see what I'm referencing as easy). I want to slide along the Y axis as it is indicated in the lower right corner (pointed to by my yellow pointer). The two faces I have chosen make perfect sense for what I'm endeavoring to achieve because the Tray will slide on top of the bottom of the Box (as shown by my orange pointers). But the Tray face and the Box face selected have the wrong orientation as per their axis indicators (pointed to by my red pointers). With the Tray, the RED line of that axis is the direction I want to move. With the box, it's the GREEN line of that axis. Currently, it is the blue line of the selected object's axis and the Z of the lower right corner axis that needs to point in the direction you want to move. Regardless of what direction you want to move, it's all about blue. No idea why it's blue, but it is. This means you have to hunt around both objects looking for two edges/faces that make sense for how the two objects get jointed. There are only a few of these... like the front of the Tray and the back of the Box. But, none of these, in this example have the blue kine of the axis pointing in the direction wanted. So you have to choose two unassociated faces/edges to joint... and then you have to painfully adjust the Placement data to get it right. That's way more work than should be needed.

[image]

Note that currently, if I select the two faces as mentioned make the most sense, they are jointed like shown next image. This isn't quite right, cuz the Tray has been rotated from its initial position (likely because of it's initial axis indicator). The two axis have been merged into one axis that shows blue lines up and down, and sure enough, if I slide the Tray, it'll move up and down. Not what is wanted.

[image]

My proposal is that you do one of two things...

1. Add an option in the Create Joint to select what axis the slide (in this case) should happen on. Then internally do the math to make that happen.
- OR -
2. When you select a face/edge to add to the joint, you are able to change the orientation of its little axis indicator, to tell the tool what orientation you expect the sliding to have, probably using the lower right axis indicator as the perspective. And the internally do the math to make that happen.

Let's discuss the logic of how I, a two week user of FC (but 3D knowledgable), thought this through while doing it the first time.
I created the Assembly, and then I added the 3 bodys to it (Box, Tray, Cover). They are added looking like the folllowing...

[image]

Notably, the original Placement of those bodys is ignored, because the src object Tray I have oriented rotated 180 degrees originally. It gets auto rotated when I add it to the Assembly. So, the next thing I do is drag the Tray to the left with the mouse. I then doube-click it to accesss its Transform and I turn it around, like this...

[image]

I then also dragged the Cover over to the right. It's oriented correctly, so I leave it as is.

I then proceed with the Slider Joint creation as discussed already, selecting the underside of the Tray, and the upperside of the Box bottom... and joint.

At this point, my expectation is that the Tray and the Box will position exactly how one would expect them to... Tray inside the box, on top of the box's bottom. That's not what happens. But, it should.

Hopefully that helps explain. The goal here, and my entire goal for FC, is that things happen that make sense. And that require as little effort on the part of the user as possible.

The current implementation of Assembly is too unexpected. Little of it makes sense... it seems mostly like guess work.

On the good side... the mouse control to move objects around. the quick access to transform, and the auto figuring out of how the two objects can be paired (though currently it may not get the orientation correct) are fantastic. These things should also be in the Parts workbench, because that's not very user-friendly either. The mistake too many developers make is that they are expert users of the tool they're developing... and they develop with the useability of their experience. That is not the majority of the users though. There's another kind of developer too... the ones that only develop, and never use the product... that's also a bad combination. :-)

luzpaz, 2024-12-05:
@PaddleStroke do you mind responding with you get a moment?

PaddleStroke, 2024-12-05:
After re-reading this again, I think this is not an issue but the user is confused about how the joints works. Which perhaps indicate that things are not as intuitive as we'd like.

But @samohd you select both faces and expect a slider joint as you want because it is what makes sense for your object. But the software cannot guess your intent. Slider joint enable movement along the Z axis (the blue one) that's it. And if you select a plane, the Z axis is its normal.

You can use the advanced offset to apply a rotation, hence moving the LCS until it match the previous X axis.

This should be closed imo. @maxwxyz if you want to double check to make sure I have not misunderstood the issue?

closed 2024-12-05 by maxwxyz

samohd, 2024-12-05:
I understand exactly how it is currently implemented. Don't force a user to have to spend a half hour trying to find two matching blue lines, pointed in the exact direction, tucked up under corners of edges, or whever they may happen to be hiding. And especially when it's not even obvious that the blue line direction is the secret to figure out. The software can absolutely be intelligent in determining the intention... or it can allow it to be made more obvious what the intention it is. It certainly does not need to rotate objects that are already correctly oriented, and force the user to be frustrated and then have to spend another half hour seeking edges that work as expected. Download my project and try it yourself. It appears the project file is no longer attached to this post... it's in my other assembly issue posts... I've made a revised version and attached here again...

[Box1b_AssemblyJoint.zip](https://github.com/user-attachments/files/18025160/Box1b_AssemblyJoint.zip)

PaddleStroke, 2024-12-05:
https://github.com/user-attachments/assets/f124cabe-634e-4000-90de-ab29618edecd

The way the slider joint works is like this. It's the same as in onshape btw.
You want the software to guess what you want. That's not possible.

samohd, 2024-12-05:
Yes, I figured out the two not obvious edges in order to achieve the sliding of the two parts as you did. What can not be seen from your video, is that for someone that didn't program the Assembly module (i think it is you that wrote this, or reworked it) it is not easy to find those two particular matching edges. Give this project to someone less experienced (without your video), and see how long it takes them to figure out those matching edges... and without them knowing about the blue axis lines being key.

As a developer of almost 50 years, I can tell you that this can be done much more intuitively. It should not be your goal to make it the same as onshape, or anything else.

There are also issues with how the assembly functions from the start of creating an assembly, with regard to adding the parts into the assembly, as I also described in this post.

While your great effort in developing for FreeCAD is much appreciated, what FreeCAD really needs on its team is someone that looks out for the beginners, and helps makes use of FreeCAD as intuitive as possible. These issues are just one example area where things are not intuitive.

samohd, 2024-12-06:
Regardless of all this, as outline in another issue, the moment you change the size of one of the objects, it'll all fall apart. So, nobody will use this until that is fixed.

ctyerman, 2024-12-13:
I don't know what the choice of the z axis based on but it to me seems both unintuitive and arbitrary.

i cant think of any circumstances where  when you select a plane you would want it to slide anything other than parallel to the plane.

I think the expected behavior should be if you select two planes that they should slide in parallel with each other, any other behavior is in my opinion just unnatural 

if the z axis if the normal to a plane then sliding would or could only happen on the x/y axis in any real system

having to try and find two specific axis that match certain conditions, rather than realistic behavior of sliding planes, seams to be like bad design

xyzzy42, 2026-03-12:
There is a way to slide in the desired direction.  It's just a pain.  What you need to do is enable advanced offset and edit the rotation for both offsets.  Rotate them 90° about the X or Y axis, depending on which axis you want to slide on (Y or X).  You need to do this for both offsets (assuming they are both planes).  That will orient the Z axis of the LCSes in the correct direction.

It would be much more user friendly if there was just a selection for "slide axis" and you could pick X,Y,Z as desired.  It could do this by making the same adjustments to the LCS for you.  But assembly features seem very secretive.  Sliders slide on the Z axis is a secret you won't find in any documentation nor would anyone explicitly say how to change the direction.

For example, the Z axis of this face as initially selected is normal to the face (and I agree with ctyerman that is virtually always the wrong axis for a plane).
[image]
I want it to slide on the Y axis (green).  So I rotate this offset 90° on the X axis to align the Z in the desired direction.
[image]
And now the Z axis of the LCS is oriented correctly.
[image]

chandradharkoneti, 2026-05-23:
I agree with the OP. When a user wants to select 2 planes to slide against each other (The name itself says SLIDER), the user must be able to select any 2 planes which the user wants to slide against each other. Finding edges along the plane feels VERY UN-INTUITIVE. Even in the popup/readout message that displays when you hover over the SLIDER Icon, nowhere does it state that the sliding ONLY OCCURS along Z-axes.

@ctyerman stated it best. Unintuitive and cumbersome. 

AT LEAST please edit the popup message to instruct the user that sliding happens ONLY along the Z-axes. I understand that when you select 2 faces, those faces can slide against each other as well as rotate relative to each other, so you need to specify an axis along which sliding should occur so that rotation does not take place, but arbitrarily selecting Z-axis is un-intuituve.

***IF changing the joint's internal code will be too complex, the SIMPLEST solution to this issue would be to PREVENT the user from selecting FACES. Only let the user select edges/axes/lines. No faces.***

And also specify that in the toast/popup message that appears on hovering over the slider joint icon.

I say prevent the user from selecting faces coz I just spent over half an hour trying to get this to work. Then came across this thread and finally understood that I shouldn't be selecting the faces, only the edges common to both faces along which the faces are to slide.