FreeCAD/FreeCAD #16259: Simple constraint dimension change causes Assembly to be totally messed up

samohd, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Here I have a simple Assembly with the tray and cover able to slide into the box. But I realize that the front edge of the box should extend out 1mm further (note the orange arrow).

[image]

So I go to the box's sketch and add +1mm.

[image]

The Assembly then becomes this:

[image]

This is a very simple example. But, anything that is changed for any body that's in the Assembly, assures having to redo the Assembly. This is irrelevant of the project. Perhaps this is being worked on, but if not...

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-03:
zip file and drag/drop into github textbox to attach, pls

samohd, 2024-09-03:
Here's a simplified example. Simple sliders configured and functioning....

[image]

Open this document... to see that view.

[Box1b.zip](https://github.com/user-attachments/files/16852354/Box1b.zip)

Double-click Sketch019, and change 37.5 to 38.5. Close the sketch.

You now see this:
[image]

Have to delete the Assembly and remake it. And this happens with just about any kind of change in bodys within an Assembly.

ialohrr, 2024-09-16:
It appears to be a toponaming problem. As you can see in provided lightly modified example, slider001 joint is between Tray001.Edge29 and Box001.Edge46 

[Box1c.zip](https://github.com/user-attachments/files/17014906/Box1c.zip)

however, after modifing the dimention on sketch19, the join no longer works and the following message error is shown in the console as Box001.Edge46 no longer exists : 

`pyException: Traceback (most recent call last):
  File "C:\Users\ianis.martin\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Assembly\JointObject.py", line 1014, in doubleClicked
    panel = TaskAssemblyCreateJoint(0, vobj.Object)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ianis.martin\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Assembly\JointObject.py", line 1334, in __init__
    self.updateTaskboxFromJoint()
  File "C:\Users\ianis.martin\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Assembly\JointObject.py", line 1637, in updateTaskboxFromJoint
    Gui.Selection.addSelection(ref2[0].Document.Name, ref2[0].Name, ref2[1][0])
                               ~~~~^^^
<class 'TypeError'>: 'NoneType' object is not subscriptable`

If the modified name of the edge/face still exists in the body but has changed location, then the assembly will show the behaviour explained in original example.

luzpaz, 2024-09-16:
> t appears to be a toponaming problem

@bgbsww is this toponaming related?

samohd, 2024-09-16:
Why would resizing involve renaming anything internal? Also note that in my example, nothing changes location in the original project by resizing.

samohd, 2024-09-16:
Also note that in addition to the tray repositioning, the cover also moves up higher. Though nothing changes in the height of the box.

bgbsww, 2024-09-16:
> @bgbsww is this toponaming related?

In terms of the project to port the toponaming code in, very unlikely.  In terms of the Assembly project running in parallel and not always getting toponaming related changes placed in it,  quite possible.

From the descriptions above, it appears to me like the assembly either has not registered itself to be informed when the contents of a link change, or is ignoring that information when told.

Is the link that is going funny the Reference1 or Reference2 ( the box1C.zip doesn't seem to have the values mentioned in those References ).

If so,  then the question becomes about PropertyXLinkSubHidden which appears to be used only by the JointObject  and CommandCreateView in Assembly, and not by anything else, which opens the possibility that it is buggy.

I'll give some thought to how we could test this.

bgbsww, 2024-09-17:
This is Assembly mangling Placement settings whenever the objects are touched.  No TNP.  Actual TNP calcs seem to be working correctly when the activity triggers them.

samohd, 2024-09-17:
I have already posted another bug (https://github.com/FreeCAD/FreeCAD/issues/16277) re Link creation not using the source object's Placement settings... and that this also affects Assembly. Could perhaps be related.

PaddleStroke, 2024-09-18:
@bgbsww assembly is using PropertyXLinkSubHidden to avoid cyclic dependencies. Because the issue is that the joint is in assembly, and the solids too. So if you try to use a regular PropertyXLink then it says there is a cyclic dependency: assembly pointing to itself.

samohd, 2024-11-28:
I've been watching the nightly builds since I posted this and other issues with Assembly... as far as I can tell, nothing has been changed in Assembly in this period. It still has all the issues I've reported. It's not usable in my opinion.

samohd, 2025-01-16:
I've given up posting issues with Assembly or FreeCAD in general, because based on them generally being closed as unimportant, I don't think there's a serious attitude towards bugs like this or others that I've posted. Which isn't to say that the efforts made by any of the developers aren't great and much appreciated, but what's missing here I think is an organized effort to fix major issues, instead of creating new ones. Over the years, I've consulted with companies mentoring developers and I've seen this kind of thing far too often. Yay, here are 12 new features... ok but it takes 10 minutes for the program to start - and there's no indication it's actually starting, and so you start it a few more times, and that its even slower to start. And this whole section you've just added 10 new features to... doesn't even work if you use anything other than your test project. So maybe we should add more features to make that one project amazing. Or well... maybe others might like to use the program too.

MiqCG, 2025-01-16:
@samohd , in my opinion it's difficult to see the bigger picture or to understand the dynamics of a big project like FreeCAD when someone (me) is not involved in a daily basis. Who knows the implications of having more than 2000 open issues at the same time managed just by a few dozens of people.
Sadly I cannot contribute in a more significant way to the bettering of FreeCAD. Let's hope there is or there will be a plan and I invite to respect the rhythms of those who are working in it.

PaddleStroke, 2025-01-17:
Following the reproduction steps and I cannot reproduce...

https://github.com/user-attachments/assets/bd32cf2b-da6c-4752-b687-28ed4005222d

Mito173, 2025-02-01:
Well, I don't know if this helps, but for me it happens even for this simple part in the attached image:
It's just a plain plate, initially with 4 holes to mount it on a surface. In the assembly, I attached a spacer under each hole and a screw on top of the holes (all fixed joints).
After the assembly was completed, I have added another 'drill hole' to the plain plate, by adding a 'circle by center' to Sketch002, including the required constraints.

After closing the sketch, the assembly is messed up and I can only delete it: 
[image]

In the following ZIP file please find my file with this assembly in its initially (still good) version:
[Board_Altitude.FCStd.zip](https://github.com/user-attachments/files/18630279/Board_Altitude.FCStd.zip)

I am using FreeCAD 1.1.0dev Revision 39980 on a MacBook with M3 chip and macOS 14.6.1

MiqCG, 2025-04-17:
@Mito173 , I have tried your files with revision 39708 and 40408 and I could reproduce the issue exactly as appears at your images but then, following a recommendation of @drwho495 , I have redone from scratch (a similar version of) the files with revision 41264, and couldn't reproduce.

Can you check if my very simplistic objects are conceptually comparable to yours and if, at your side, the issue is not present anymore with the last revision?

[Assembly rev41264.FCStd.zip](https://github.com/user-attachments/files/19798288/Assembly.rev41264.FCStd.zip)

Mito173, 2025-04-19:
Hi @MiqCG , I think your drawing is comparable to mine. Since you did not seem to have the issue with your drawing in revision 41264, I also tried with same revision 41264 on my MacBook and still had the issue. 

Since your had a different result, I now installed FreeCAD revision 41264 on a Windows 11 computer and did some tests. The resullts may help to get a step further:

- If I create the initial part for the assembly on the **Mac**, and add another hole to it on **Win11**, the assembly will be screwed up.
- If I create the initial part for the assembly on the **Win11**, and add another hole to it on **Win11**, the assembly is **OK**!
- If I create the initial part for the assembly on the **Win11**, and add another hole to it on **Mac**, the assembly is also **OK**!

So in my opinion, the issue is not in the Assembly itself but things go wrong much earlier, when creating the sketch with the holes, used in the joints of the assembly - and, it only happens on MacOS!

In the following please find the steps I did for this sketch, which seemingly caused the issue:

01. Create new Sketch on top surface of Pad
02. Add 4 times 'Create circle by center'
03. Chose 'Constrain horizontal distance'
04. Click on middle dot of 1st circle
_Debug errors are shown from this point, but they occur on Mac as well as on Win11:_
_10:20:57  <Sketch> SketchObject.cpp(11811): invalid subname ;g1v1;SKT.Vertex1
10:20:57  <Sketch> SketchObject.cpp(11811): invalid subname ;g1v1;SKT.Vertex1
10:20:57  <Sketch> SketchObject.cpp(11811): invalid subname ;g1v1;SKT.
10:20:57  <Sketch> SketchObject.cpp(11811): invalid subname ;g1v1;SKT.
10:20:57  <Sketch> SketchObject.cpp(11811): invalid subname ;g1v1;SKT.Vertex1
10:20:57  <Sketch> SketchObject.cpp(11811): invalid subname ;g1v1;SKT.Vertex1_

05. Continue completing horizontal constraints for the two upper circles
06. Constrain vertical distance for the two left circles
07. Constrain orizontal for the two right circles
08. Constrain vertical for the two lower circles
09. Constrain diameters for all 4 circles (4 millimeter)
10. Close fully constraint sketch
11. Use Pocket tool on just created Sketch001, type 'Through all'
---
12. Add isometric fastener with diameter M4 and length 12 millimeter
13. Create Assembly, insert Body-part and 4 times the fastener
14. Create Fixed Join like displayed in picture for all 4 holes/screws

I performed the exactly same steps to create the file on Mac as well as on Windows. Please find attached ZIP FreeCAD_16259_20250419.zip with two versions of the same drawing with status as of step 12 - one time created on Mac, one time on Win11, same FreeCAD revision 41264.

[image]

[FreeCAD_16259_20250419.zip](https://github.com/user-attachments/files/19820466/FreeCAD_16259_20250419.zip)

drwho495, 2025-08-12:
> Here's a simplified example. Simple sliders configured and functioning....
> 
> [image]
> 
> Open this document... to see that view.
> 
> [Box1b.zip](https://github.com/user-attachments/files/16852354/Box1b.zip)
> 
> Double-click Sketch019, and change 37.5 to 38.5. Close the sketch.
> 
> You now see this: [image]
> 
> Simplest solution is to then delete the Assembly and remake it. And this happens with just about any kind of change in bodys within an Assembly.

errors in this model are caused by a missing reference (at least a reference that should be triggered as missing)
you need to redo the last slider joint and/or wait for a pr to get merged

drwho495, 2025-08-21:
> Can you please elaborate @drwho495 ? I cannot replicate the issue at all and don't see bad references.
> And what PR are you talking about?

Are you on the latest daily build? The pr (I think) I was referring to is merged now (it was the one that makes the property links update more often)

The part design fix I implemented that caused assembly issues could also be related to this

drwho495, 2025-08-21:
ok i tested it again and everything works as intended

don't use the latest daily build, there is an annoying regression, but (hopefully) the next build will have the fix for that
that build will be ready to use with this project

MiqCG, 2025-12-31:
Hi,
I have tested the files Box1b, Box1c, Board_Altitude and my own Assembly rev41264
with 1.1.0rc1 and there wasn't any broken assembly after modifying sketches. 
I think that this issue can be closed.

Happy new year

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow) AppImage
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Catalan/Spain (ca_ES)
Navigation Style/Orbit Style/Rotation Mode: Revit/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.8695/1.07292
Installed mods: 
  * StructureTools 0.0.1
  * sheetmetal 0.7.24
```