FreeCAD/FreeCAD #16261: window temporarily hidden when opening first document, status bar gets hidden (qt6 only)

adrianinsaval, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/user-attachments/assets/c30eda0c-eaff-4d10-804f-56db73cf989a

https://github.com/user-attachments/assets/1e8f06c9-1a60-48af-ba42-5d88114b44c4

see videos, the issue seems to be exclusive to qt6 build. It happens only for the first document that is opened

Also when using the start page button the application stays in the start page tab instead of going to the created document

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * fasteners 0.5.27
  * Curves 0.6.45
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
  * AirPlaneDesign
  * sheetmetal 0.4.24
  * OpenTheme 2024.8.17
  * Ondsel-Lens 2024.7.5.02
  * CfdOF 1.27.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-09-03:
@xtemp09 this issues is somewhat similar to the light source preference page making the window dissapear, maybe you have a clue on what's going on here?

xtemp09, 2024-09-03:
> this issues is somewhat similar to the light source preference page making the window dissapear

chennes found that _showEvent_ was invoked twice.[^1] The PR #15779 fixed a crash, keeping the double call of the _showEvent_ function. My PR just straightened everything up.

> Also when using the start page button the application stays in the start page tab instead of going to the created document

I think the issue #15491 and the pull request #15944 may be connected.

> maybe you have a clue on what's going on here?

@adrianinsaval, I cannot reproduce, so I don't know. Focus issues, layout issues, theme issues.

[^1]: https://github.com/FreeCAD/FreeCAD/issues/13357#issuecomment-2269210258

xtemp09, 2024-09-04:
@adrianinsaval, since you are able to reproduce the problem, could you use _GammaRay_[^1] to inspect the events?

[^1]: https://github.com/KDAB/GammaRay

kadet1090, 2024-09-05:
> @adrianinsaval, since you are able to reproduce the problem, could you use _GammaRay_[1](#user-content-fn-1-c2e893fc95641d68734153bfd6e586a1) to inspect the events?
> ## Footnotes
> 
>     1. https://github.com/KDAB/GammaRay [↩](#user-content-fnref-1-c2e893fc95641d68734153bfd6e586a1)

I can reproduce the issue and have gammaray, can you point me what to look for?

xtemp09, 2024-09-05:
> @xtemp09 I can reproduce the issue and have gammaray, can you point me what to look for?

The _showEvent_.

closed 2024-09-12 by wwmayer