FreeCAD/FreeCAD #16262: Stylesheet does not get applied / selection not visible

maxwxyz, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the latest weekly the colors are changed so you cannot make out which elements are selected. Maybe stylesheets are not applied correctly.

FreeCAD Light (2 selected edges):
[image]

FreeCAD Dark (1 edge selected, 1 preselected):
[image]

### Full version info

```shell
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-03:
I'll label this as blocker as you cannot see the selection.

luzpaz, 2024-09-03:
Should we start adding what Qt theme is being used in the About info ?

maxwxyz, 2024-09-03:
yes and maybe Stylesheet?

MisterMakerNL, 2024-09-03:
I can confirm this.
I'll make a quick pr.

closed 2024-09-09 by yorikvanhavre