FreeCAD/FreeCAD #16264: New SplashScreen only gets displayed for a fraction of a second.

GAZ082, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Looks like all the loading gets performed and after that the splash appears for a blink of an eye.

https://github.com/user-attachments/assets/ed066d97-e502-4a78-b62a-e463d5653db0

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.8.17
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-03:
Confirmed, I re-enabled the messages just so I could see the SplashScreen and I'm using a 12 year old box so what it's like on a new piece of kit, must be close to invisible. A minimum length of display time of say one second wouldn't be unreasonable. @qewer33 did you not experience this scenario in testing?

FEA-eng, 2024-09-03:
For me, it's working normally but I still have 0.21.2 and rather old computer.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

qewer33, 2024-09-03:
Hmm yeah, it's because my PR disabled the splash messages which was unnecessarily taking time but ig it's way too fast on modern computers now... (On my machine it does take a second though but sometimes a little less)

So we have a couple of options:
- Bring back the splash messages
  - I gues we can do that...
- Set a fixed delay
  - Terrible idea *and also the internet would meme it to hell and back if it somehow gets discovered*.
- Ensure the splash screen shows for a minimum X amount of time, implemented via a timer or something
  - Still not very good but certainly a better idea than the last one since it'll be more complex and normal people won't be able to understand and meme it.
- Actually fix the scuffed startup routine, for some reason the splash screen shows a few seconds after FreeCAD starts and hides a few seconds earlier than when the main window appears. Also the only things that gets done while the splash is showing seems to be GUI initilization which as we can see now with the splash messages disabled, doesn't even take that long...
  -  Now the problem with this idea, we will need to mess with the startup routine and the base application code so it might cause a lot of platform specific bugs, not a very good idea to implement before the 1.0 release.

The third option seems like the best one but I'm open to all ideas. Please express your thoughts.

GAZ082, 2024-09-04:
> This is really good, the quicker it gets to open, the better it is.

In theory. But in this case, the WBs and stuff gets loaded before the splash screen. So you are kept staring at the desktop, wondering what's going on and then boom.

3x380V, 2024-09-06:
No, WBs are loading while splash screen is shown, just see the code. With #16334 you can have messages back, so just store time splash was shown, compare with time it is about to disappear and eventually extend it. That's possible, since once Qt message loop is running, timers work.

GAZ082, 2024-09-06:
@3x380V 

With "messages" you are referring to the notification that something is being loaded? Ie, the workbenches?

3x380V, 2024-09-06:
@GAZ082, see https://github.com/FreeCAD/FreeCAD/blob/fe40748c10503eed367840239db3e0f81873e359/src/Gui/StartupProcess.cpp#L430-L434
You might as well move showing splash earlier in the startup process, I just tried to put at the beginning of
https://github.com/FreeCAD/FreeCAD/blob/fe40748c10503eed367840239db3e0f81873e359/src/Gui/StartupProcess.cpp#L218-L220
But you need to check for side effects. Also OS needs some time to get image into memory and there is nothing application can do about.

leoheck, 2024-09-06:
> In theory. But in this case, the WBs and stuff gets loaded before the splash screen. So you are kept staring at the desktop, wondering what's going on and then boom.

I see, however on my machine, the launch of Freecad is really quick now since 2 or 3 weekly builds ago.

3x380V, 2024-09-06:
We want do show splash only if `Gui::Application::runApplication()` not `Gui::Application::runApplication()` is about to be executed and only if it is not executed from Python. Is that correct?

3x380V, 2024-09-07:
Please build and test #16334. Whatever is written in that PR comment still apply.

mnesarco, 2024-09-20:
Please consider my PR for 1.0, it is quite small and works until a bigger redesign of StartupProcess:
https://github.com/FreeCAD/FreeCAD/pull/16685

GAZ082, 2024-09-26:
I can confirm that this is happening in the 1.1 branch, not the 1.0.RC

luzpaz, 2025-12-04:
This is still happening, btw

GAZ082, 2025-12-04:
I like the splash, but if it's so hard to make it work as it should, do yes, let's scrap it.