FreeCAD/FreeCAD #16266: [Feature] Assembly: No way to differentiate between activated joint (true or false)

MiqCG, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a Feature request to add some visual clue, at least at the tree, to distinguish between an active joint (Activated = true in the Properties > Data > Joint section) and a deactivated one. 
I think that is useful, if not essential, to being able to keep up (easily) when you have a decent amount of joints and you are testing the behaviour of an assembly in different situations.
Maybe an specific icon or a strikethrough line on the name of the joint, I don't know.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * A2plus 0.4.66
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MiqCG, 2024-09-03:
PS: on the other hand, I think it is not obvious that exists the possibility to disable a joint. I found it by chance and because it was strange that no issue was created for their existence. Maybe it can be easier to find if it is also an option of the drop-down menu of the Task that appears on double click to a joint.

MiqCG, 2024-11-03:
Reading the FEM Release notes of 1.0*, I have seen that the new feature of Supressing constraints ( [PR #12359](https://github.com/FreeCAD/FreeCAD/pull/12359) ) is indicated with a strike-through line
[image]

and I thought it can be as well a good way to indicate in the Tree that an Assembly joint is inactive.

@maxwxyz , when you consider a good time, I think it is worth to call design-working-group and/or PaddleStroke for their vision.

* https://wiki.freecad.org/Release_notes_1.0#FEM_Workbench

maxwxyz, 2025-01-02:
I think the proposal is fine, although if you have many joints in the assembly and just see Slider001 and ~Slider 002~ I'm not sure if you know what is suppressed in the assembly itself. Maybe it needs also some hints in the 3D view of which joints are active.
What happens in the assembly if the joint is not active? Is it just as if there was no joint at all?
@FreeCAD/design-working-group please share your opinion

MiqCG, 2025-01-02:
> What happens in the assembly if the joint is not active? Is it just as if there was no joint at all?

Yes, if the assembly joint is deactivated, it is not taken into account in the solving of the assembly and the joint is free to move.
I agree with you that an indication of the non-active state of the joint/s in the 3D View is also necessary. For example a joint marker with dimmed colours or axis in grey.

luzpaz, 2025-01-15:
cc @PaddleStroke https://github.com/FreeCAD/FreeCAD/issues/16266#issuecomment-2568178737

PaddleStroke, 2025-01-15:
Strikethrough is a good indication for the tree. And for the view we can gray the marker. This way a deactivated joint can be seen both in the tree and in the view.

maxwxyz, 2025-01-15:
maybe it can be done with the Suppressed feature? Suppressed items are already strike through and maybe the same command can be used for joints, so similar concept for the user?

PaddleStroke, 2025-01-16:
Ah yes you are probably right.

pierreporte, 2025-01-16:
> What happens in the assembly if the joint is not active? Is it just as if there was no joint at all?

It would be like the joint doesn’t exist.

> Maybe it needs also some hints in the 3D view of which joints are active.

Last year I suggested to @PaddleStroke to draw a line between the two JCS that are visible when the joint is selected. That line could be hidden when the joint is deactivated or suppressed. If this line doesn’t become dashed when behind geometry, it could be made dashed for this purpose so the link is more obvious. Though there will be a problem if they are coincident, which can happen if the suppression is not followed by movement of the parts. Maybe adding an icon?