FreeCAD/FreeCAD #16271: [Draft] Using Rotate on a heavy model (2163 objects) is approx 40 times slower in current 0.22dev than 0.21.x

Syres916, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In comparison to a heavy model with only 5 objects but with over 100,000 vertices, it's the number of objects in the document that causes the slowdown.

I'm using only keyboard input to carry out the _Draft>Rotate_ (eliminating any mouseover/hover slowdowns) after selecting the root object. So with a Centre of _0,0,0_ and an initial angle 0f _10_ degrees and a final angle of _50_ degrees, in  0.21.x the command is completed in under a second. Using 0.22dev it takes 40 seconds and mine isn't slow compared to the recent forum [report](https://forum.freecad.org/viewtopic.php?p=779879#p779862) where a user clicked the first point for the centre and it was still going for an hour before abandoning the process.

There is an error raised in the Report View:

`21:07:03  4.3e-08 <App> Document.cpp(2800): Recursive calling of recompute for document document_name` 

I presume that _Draft>Rotate_ is just one of a few commands that will be affected but I've yet to prove that.

### Full version info

```shell
Last good build I have is:

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage (Mid May 2024)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)

First bad build I have is:

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage (end June 2024)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-03:
I'll carry out a Git Bisect tomorrow and get a commit reference.

Syres916, 2024-09-04:
Update: as usual there's more to this than just one thing, firstly the initial builds I had to show Good and Bad were both AppImages and the self compiled Bisecting is being done on a new distro Linux Mint 22 so not comparing Apples with Apples.
There is a slowdown using self compiled builds (half that of the AppImage) and at present I've got it narrowed down to:

Good:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37907 (Git) 25th june 2024
Build type: Release
Branch: (no branch, bisect started on main)
Hash: e49f74aed985de7dea86e8272e4aeece13fe9c97
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
```

Bad (but not as bad as the AppImage):

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37917 (Git)
Build type: Release
Branch: (no branch, bisect started on main)
Hash: 8def1e94dcee744ae4178964cb485c9e437592fa
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```

Then it occurred to me that there's an issue in existence for appallingly slow changing of Theme using AppImage. Well that took me a considerable length of time to track down as it's been moved to the Bundle Repo https://github.com/FreeCAD/FreeCAD-Bundle/issues/302 and when I look at the report date with the build numbers quoted, I don't believe in coincidences.

If no-one can get to the root of the AppImage side of the slowdown then we might have to recommend that users extract the Release build to a folder as that was proved to mitigate _some_ of the performance impact. Meanwhile I'll get the last commits bisected later this evening.

Syres916, 2024-09-04:
@bgbsww of the 40 seconds slowdown about a quarter is due to this [commit](https://github.com/FreeCAD/FreeCAD/pull/14898/commits/3c3709cb5d0173eaa0519996e5c688752ffbca34) , is there some advice we should be putting into the [Release Notes](https://wiki.freecad.org/Release_notes_1.0) and/or [Known Issues Feedback Topic](https://forum.freecadweb.org/viewtopic.php?f=3&t=30574) with regard to opening pre1.0 files with a large number of (over 400) objects, for example:

_It is recommended to save the file as a new name such as ....1_0.FCStd so the TopoNaming Algorithm can create the relevant new data before starting any operations on the document file without risking any damage to the original file if it's required to be opened by 0.21.x in future._

I'll carry on tracking down the selection by mouse issues tomorrow which go back at least 500 commits.

bgbsww, 2024-09-04:
I'm a little confused here.  Any chance that commit reference is wrong?  That commit just enables loading and saving of element maps, so it's hard to see how it would be affecting the rotate.

However, I can certainly see a case where we might want to encourage recomputation of files without element maps for many reasons.  In fact, we might even want to force that on file opening (Yes, I know, UI change, we'd need a special exception, etc, but if this is affecting overall behavior and performance, that could need doing).

Syres916, 2024-09-05:
> I'm a little confused here. Any chance that commit reference is wrong? That commit just enables loading and saving of element maps, so it's hard to see how it would be affecting the rotate.

Yep, I'm completely happy that it's the correct commit reference. I've just tried saving the file under a new name using the build info below and then restarting FreeCAD and loading the new file with element maps present and it's just as slow to complete the rotate.

Initially tomorrow I'm going to concentrate on the mouse selection issues but then I'll add some crude debugging to the code in the commit above to give you an idea where I think the impact is being felt the most. Fortunately on the AppImage build from the end of June I can actually see the each top level object for the main branches being recomputed, it's that bad which is why my thoughts were _Tree_ or _Document_ being the place to investigate.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38641 (Git)
Build type: Release
Branch: main
Hash: 01d9e0c093ed35de24b45194040069a950132045
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66

```

bgbsww, 2024-09-05:
I think what you are trying to say here is that the saving and restoration of element maps ( as opposed to not having any element maps) is triggering some kind of slow down in draft>rotate, and a symptom is the recursive calling of recompute warning.  I can buy that.

However, that would point at something else that this commit is triggering - essentially that there is a problem with rotate that is triggered by the presence of an element map, not that there's an issue with this commit.

I think we need to be concentrating on if a file with element maps is creating that recursive calling issue, which would be independent of the commit in question.  Thus my confusion.

If you load the file in a version of freecad from a commit prior to this one, and then force a full recompute to create the element map, does the rotate then slow down?

bgbsww, 2024-09-05:
Can't seem to trigger that Recursive Calling warning with trivial examples - is there a file that will cause it to happen?   Or a stack trace of when it happens - I'd love to know what's calling it?

Syres916, 2024-09-05:
> I think what you are trying to say here is that the saving and restoration of element maps ( as opposed to not having any element maps) is triggering some kind of slow down in draft>rotate

No, sorry if I wasn't clear, it doesn't make any difference if I open a 0.21.x version of the file or one saved with the dev build quoted above, using _Draft>Rotate_ is equally slow. But, and I've only just tested this, if I then use _Draft>Rotate_ a second time to either put the model back to it's original rotation or even rotate it further then it executes at full speed.

Now if I use the Property Editor and just change the Angle to achieve the exact same result immediately after loading either of the files it's instantaneous so whatever's going on is only triggered from the Python code in Draft Wb. I can't publish the file on here as it's not mine but give me another day to do some digging and if needed I'll have to mail you a copy.

bgbsww, 2024-09-05:
Ah, thanks.  I'll wait for your further info and we'll take it from there.

Syres916, 2024-09-05:
@bgbsww I've sent you a forum PM with updated info and FC model location.

bgbsww, 2024-09-06:
Thanks, @Syres916.  Ok, so first off, the loading of the FC model is definitely agonizingly slow, and that is because of a very large file, which also issues many "Pending recompute for generating element map:" warnings and the corresponding code is executed.  This is the code that while already present, first gets enabled by the PR in question.

That corresponding code is `owner->getDocument()->addRecomputeObject(owner);`  which ultimately does an `obj->touch();` which results in having a large number of objects that need to be recomputed - 427 in the example file case.

It is this recompute that appears to be taking the time - not the Draft->Rotate.  Once the recompute has happened, which incidentally the Draft->Rotate appears to trigger, thus the first time slowness, then all seems normal.

I see four possible solutions to this:
1) Copy the LS3 approach of putting up a window recommending/forcing a recompute on loading a file like this.  That's a UI change :(
2) Silently just do the recompute; or add a config property permitting the forced silent recompute.
3) Add a bunch of documentation that nobody will ever read (okay, that's a little bit of my bias showing).
4) Remove / disable the code that does the touch(), and just let people work with an unrecomputed model.  This might lead to disabling TNP mitigation later, but I don't have an example to prove that.

Syres916, 2024-09-06:
Yep I agree with you regarding solution option 1, thanks for the investigation work.
I'll carry on with the mouse selection work and hopefully there won't be a huge time gap between version 1.0 and 1.01 as I think it'll be a big step to a more stable/useable app.

Syres916, 2024-09-18:
I'm closing this, thanks for https://github.com/FreeCAD/FreeCAD/pull/16540 now merged.

closed 2024-09-18 by Syres916