FreeCAD/FreeCAD #16272: Triage: Add Stylesheet and Qt theme being used to About info

luzpaz, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Given the  
1. amount of [Stylesheet](https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Topic%3A+Stylesheets%22) bugs there are 
2. confusion around Qt's internal themes (https://github.com/FreeCAD/FreeCAD/issues/15868)

It would be helpful for bug triage to have the option to see what stylesheet and Qt theme are being used. The proposal is that when one copy/pastes their [About](https://wiki.freecad.org/About) info there could be a line that shows both of these as consolidated: 

`Stylesheet/Qt theme: OpenTheme vX.Y.Z/Fusion`  

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Qt theme: OpenTheme vX.Y.Z/Fusion
```

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Core

### Anything else?

Originally discussed in https://github.com/FreeCAD/FreeCAD/issues/16262#issuecomment-2327131495

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-09-03:
Also information about high DPI and scaling percentage would be good

luzpaz, 2024-09-04:
@kadet1090 great idea! Any idea what functions we could use to ascertain these variables ?

luzpaz, 2024-09-05:
@benj5378 thank you!

> Also information about high DPI and scaling percentage would be good

Any chance you can also add this?

kadet1090, 2024-09-05:
https://stackoverflow.com/questions/20603931/how-to-get-the-dpi-of-the-display-screen-in-qt - this can be used to obtain the required value. We can simply use the DPI value, does not have to be scale factor in percents.

luzpaz, 2024-09-06:
Should I open a new ticket for adding 'high DPI and scaling percentage' info ?

benj5378, 2024-09-08:
@luzpaz I will try to take a look at it tomorrow or Tuesday

closed 2024-09-08 by chennes

luzpaz, 2024-09-08:
Thanks @benj5378 :pray:

luzpaz, 2024-09-09:
> Should I open a new ticket for adding 'high DPI and scaling percentage' info ?

@benj5378 kindly made a PR for DPI in #16390  
Bravo!

furgo16, 2024-09-09:
Thanks for your work. I filed a follow-up issue with a suggestion to consider: https://github.com/FreeCAD/FreeCAD/issues/16391

PR #16281 (2024-09-08): [Gui] Add style, theme and qtstyle info to clipboard about info
Fixes #16272

```
Stylesheet/Theme/QtStyle: Behave-dark.qss/FreeCAD Classic/Fusion
```

For testing, you might have to manually copy the changed preference packs
commit 33e8bf61 (2024-09-08): [Gui] Add style, theme and QtStyle to clipboard about info

Fixes #16272
commit 8cdcbe36 (2024-09-08): [Gui] Add style, theme and QtStyle to clipboard about info

Fixes #16272