FreeCAD/FreeCAD #16274: Error with Pad up to face along custom direction

AustinPackard, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The error message: "SketchBased: Up to face: Must not be parallel to extrusion direction!" displays when making a Pad feature with a custom direction.
This seems to be a regression from 0.21.2. Seemingly related issues: #7879 #7880 

Steps to reproduce:
1. Create a closed profile on the XY-plane that is entirely left of the Y-axis.
2. Create a sketch on the XZ-plane containing a line that has a positive slope (not vertical).
3. Create a Pad using the sketch from step 1.
3.1 Set Type to "Up to face" and select the YZ-plane.
3.2 Set Direction/edge to "Select reference..." and select the line created in step 2.
4. Error "SketchBased: Up to face: Must not be parallel to extrusion direction!" takes place.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.7.5.02
  * OpenTheme 2024.8.17
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Pad up to face error.zip](https://github.com/user-attachments/files/16857884/Pad.up.to.face.error.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-04:
@AustinPackard could you provide a demo file?
@FlachyJoe FYI

AustinPackard, 2024-09-04:
Demo file "Pad up to face error.zip" provided in the "Anything else?" section above.

AjinkyaDahale, 2024-09-17:
Macro to reproduce:
[bug-16274.zip](https://github.com/user-attachments/files/17032866/bug-16274.zip)

PR #16539 (2024-09-24): [PD] Fix Pad uptoface with custom direction
Fix #16274
closed 2024-09-24 by chennes