FreeCAD/FreeCAD #16276: [BUG][UI/UX] Stacked panels 'unstack' in overlay mode.

obelisk79, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

wmayer, last year, modified panel behavior in place of the old 'Combo View' and allowed stacking of different panels for more customization options. This functionality works quite well, and is appreciated. However when attempting to do this while in overlay mode, it is impossible. Even when stacking Task and Property panel for a tabbed combination, it will tile these panels when switching to overlay mode. Screenshots to show before and after:
Before (Task and Property stacked)
[image]

After (only activated overlay)
[image]

Switching back to non overlay results in non-stacked panels
[image]

I believe this is a bug because it is reasonable to expect switching between overlay and non-overlay to not change other aspects of panel configuration. I wouldn't consider this any kind of show-stopper, but consistency is beneficial for overall user experience, I presume the issue lies within the overlay manager as it was developed before the changes to the combo view / panel stacking behavior.

### Full version info

```shell
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) AppImage
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2025-07-19:
Still valid

FEA-eng, 2025-09-20:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-02-22:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-08-25:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260720 (Git shallow)
Build date: 2026/07/20 16:51:42
Build type: Release
Branch: grafted,grafted
Hash: 057d51f846f535f4f0b48492e586be0a76b9efe2
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```