FreeCAD/FreeCAD #16277: App::Link Make Link incorrect orientation * Affects Assembly adding of Body as a Link too

samohd, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached project select Insert > iBase object.  Link actions > Make Link.

Result will have the link object in the wrong orientation as shown here.

[image]

[Box1b_MakeLink2.zip](https://github.com/user-attachments/files/16860788/Box1b_MakeLink2.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-04:
This is intended behavior. If you want the Link to follow the Placement of the original object you should change "Link Transform" to true. In the given example it is better to create a Link to the Body.

IMO this can be closed.

luzpaz, 2024-09-04:
Closing per Roy's feedback

samohd, 2024-09-04:
Unexpected behavior should never be intended behavior. Why make more work for the user? The Placement should stay as is. My understanding is that this is the case with Clone and Shape Binder. The whole point of a Link, is that's its a pointer to the src.... so everything should be the same. This also causes a lot of problems with Assembly.

Also... linking to the Body, doesn't work... because following actions in the dst body don't take affect. For example, I can't make a pocket in the linked Body in the dst body. This only works the way I instructed.

Roy-043, 2024-09-05:
What a user expects depends on their previous experience. The Link is intended as a cloned copy. In most cases that clone should appear in a different location from its original.

It is not clear what you want to achieve. It may help if you start a forum topic and explain the whole workflow you have in mind. If required this issue can be reopened later.

samohd, 2024-09-05:
All other clone methods put the copy in the same place. What happens here is not just shifting it to the left or right so you can see where it is after its cloned, it's changing the axis on which it exists in the source. Which makes little sense for a copy to want to be done. I have already been discussing Links in the forums. Nobody responding knows anything about them, and just tells me to use Draft Clone or Shape Binder or Sub variations. Even the GUI hover over Link icon in toolabr tells you that they are more efficient. And as an actual Link... they should be.

It seems weird to me that something deemed worthy of its own space on the Toolbar that's there all the time in Part Design and related views, has such problematic functionality. And from what I can tell, Link has been around for some years.

Roy-043, 2024-09-06:
OK, I'll reopen this.

samohd, 2024-12-06:
"we’ll close this issue automatically in 20 days to keep our backlog tidy"  Someone should turn off this dumb feature. If we haven't fixed it yet, and nobody reminds us of it, we'll just forget about it and pretend our iisue list doesn't exist. Good idea!

closed 2025-08-02 by github-actions[bot]

luzpaz, 2025-08-02:
cc @PaddleStroke are you aware of this ?

FEA-eng, 2026-02-09:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260204 (Git shallow)
Build date: 2026/02/04 19:46:02
Build type: Release
Branch: (HEAD detached at 27587b9)
Hash: 27587b9cd2bc8d686187149feee7b123a65b8c77
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```