FreeCAD/FreeCAD #16278: Draft/Bim: The problem of summing displacements when transforming objects built on the basis of simpler forms (sketch/line)

kaiwas, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I first started studying the program, I immediately encountered the problem of models exploding with different transformations.
The problem manifests itself depending on the selection of objects.
For example, consider a wall based on a sketch and a window.

If I move the wall object (only the wall is selected) - I get the correct result. At the same time, the basic sketches remain in their previous place.
If I move the wall together with the sketch - a double move occurs, since the object transformations and the sketch transformation are summed up.

When rotating, everything gets worse (as you can see in the video - the window is distorted when rotating)

This system is very inconvenient, especially if you need to move a lot of objects at the same time, some of which have parent elements, while others do not. You have to very carefully monitor what is selected in the tree or in the tree view.

https://github.com/user-attachments/assets/faeb477a-6849-4a4b-94f4-38f11ea3b394

Of course, there are workarounds for this problem - you need to know exactly what will happen during the transformation and, most importantly, what exactly should be selected.

In addition, the preview of the result and the result itself are very different.

A small discussion of this problem on the forum.

https://forum.freecad.org/viewtopic.php?t=90184&sid=6aee21b3a48d3471433aa357593c3594

I see a possible solution that can be implemented like this.
When we have an object + base sketch selected, the transformation should be done only for the sketch.

Perhaps this approach has its drawbacks. But the fact that the current system is very unclear is an absolute fact and it needs to be changed somehow. Let's open a discussion and try to find the best solution to this problem.

Attached is a file for testing. Remove the zip extension.
[Move and rotate.FCStd.zip](https://github.com/user-attachments/files/16862382/Move.and.rotate.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-04:
Agree. As much as the sketch-based system is powerful, it is also inconvenient in many cases because of that multiple transforms. Therefore, we need windows (and walls) that can also NOT be based on sketches. So simple windows have no underlying dependency, and the sketch system is still there for people who need something more powerful.

That's part of my nativeIFC roadmap at https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/roadmap2024.md It has been on hold because of the imminent 1.0 release, but we'll resume development soon.

kaiwas, 2024-09-04:
> That's part of my nativeIFC roadmap at https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/roadmap2024.md It has been on hold because of the imminent 1.0 release, but we'll resume development soon.

Okay. I'm very glad that changes are planned.
Unfortunately, I don't have programming skills (I would have joined the development long ago), but I will be happy to spend as much time as necessary to analyze the most convenient methods and to test.

I understand that it is not advisable to make global changes to the approaches at this stage (we are all eagerly awaiting the release of 1.0).

But what do you think - could we at least change the preview algorithms? (If this is really difficult, then we can postpone this task)
After all, if the preview showed the real result after executing the command, it would already be very convenient. After all, the user would see in the preview that his actions will lead to a breakdown and would simply cancel the action.
I don't know how this is done at the code level, but it looks like the preview does not perform the summation of movements.

kaiwas, 2024-09-04:
In fact, the current sketch-based version is quite functional. And even convenient for some tasks.
But there is an obvious problem with selecting elements.
If there was a way to select everything only in the tree view, or in the 3D window (something that would not lead to double transformations), then perhaps nothing would have to be changed.

But now, to select a window, you must click on it. (This, by the way, is also related to the problem of views).
https://github.com/FreeCAD/FreeCAD/issues/15983

Some new selection commands or filters are needed.

Maybe it would be easier to implement an improved selection than to change the entire sketch-based concept to another.

yorikvanhavre, 2024-09-04:
Have you tried playing with the "Move Base" property? That might help.

Unfortunately I have no clear idea of how to solve that problem immediately, without the refactor I explained above...

kaiwas, 2024-09-04:
Yes, with this setting it gets a little better.
But again, everything depends heavily on the choice.

If you select only the wall, the window no longer moves (without this setting, it moved).
If you select all objects except sketches, then everything moves perfectly to the desired distance. The model does not fall apart.
If you select all objects together with sketches, then the movements are summed up (instead of moving by 1 meter, you move by 2 meters). The model does not fall apart.

https://github.com/user-attachments/assets/f00eab72-5022-4c81-ac90-46fb93d1923e

By the way, I noticed that the color of the sketch lines, which I set to red, at some point changes to black.

kaiwas, 2024-09-04:
It seems I have an idea that would allow not to change the concept of sketches much, but would preserve the integrity of the models during transformations.

When the global setting "move base" is enabled by default, the properties responsible for positioning in the object built on the basis of the sketch should be either locked (zero. and displayed in gray). This setting is active by default and can be unlocked if necessary (some button with a lock).
When we move or rotate this object (the sketch does not necessarily have to be selected) - the transformation data is transferred to the sketch level and it is the sketch that moves.
As a result, our sketch always moves, and the object moves after it.

Visually, we transform the wall, but in fact, the position of the sketch changes. We can also move the sketch itself.

But we have a problem with link objects. They have their own positioning, which does not always coincide with the sketch. For such objects, the positioning properties are automatically unlocked at the moment of creation and we can move the link object as usual.

[image]

What does this give us? This will eliminate parasitic double transformations. At the same time, having unlocked the properties, we can return to the current algorithm.

kaiwas, 2024-09-04:
I simplified the picture with the concept a little. But I will also leave the old one. In short, this can be described by the phrase "substitution of parameters"
[image]

yorikvanhavre, 2024-09-04:
Indeed if you select both the window and the underlying sketch, and you move/rotate, you are explicitly applying the move to both the sketch and the window. We could introduce a filtering system that removes the sketch from the selection if the host window is selected too and do that for all BIM objects, but it's error-prone and will require careful testing, so it is not wise to do it before 1.0

kaiwas, 2024-09-04:
> to do it before 1.0

I understand all this. Such tasks are not solved on the first try in one evening.
Therefore, I am not inclined to rush anyone on any of my issues. If they are solved within 10-20-50 years, that will also suit me)
Maybe my descendants will have a convenient tool for work.
And I already have everything I need for work. I am simply trying with my "whims" to provide grounds for improving the program that I have already come to love (along with all the developers).

paullee0, 2024-10-19:
@kaiwas  mention this in the above discussion.

If fact, with SketchArch add-on, the whereabout of their base Sketched does not matter, and do not interfere.  This is one 'problem' the SketchArch is developed to tackle :)

Personally, I put all windows' sketch at (0,0,0) or close to it, or at an array layout, so it is easier for 'comparison' - but their placement do not interfere the Window's and their Link's Displacement.

paullee0, 2024-10-19:
Or if without SketchArch, maybe a property similar to Link Transform in link object could be adopted in Arch Objects?

So Arch Objects' location is 'de-linked' and no more messing up ?

paullee0, 2024-10-27:
> When rotating, everything gets worse (as you can see in the video - the window is distorted when rotating)
> 
I have another at @kaiwas and note this - that's why I advocate the Normal of Window should be set to Auto (0,0,0) for extrusion, that would fix a great portion of the problem.

Then, the whereabout of the Base object (Sketch) of the Window shouldn't matter, as I implement in my add-on.

The Move Base property could be retired.

Any other idea? :)