FreeCAD/FreeCAD #16284: BIM: profile always created on XY plane regardless of working plane

furgo16, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Switch to BIM Workbench
2. Set the working plane to "Side (YZ)"
3. Click on the Profile command
4. Select a profile preset in the Task window
5. Click on a point on the 3D window where the profile should be created
6. **Expected**: the profile is created on the YZ (side) plane
7. **Actual**: the profile is created on the XY (top) plane

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38635 (Git) Snap 1125
Build type: Release
Branch: main
Hash: 7274dac185339a45020f1dc91f2b6c7478b2b0aa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.30
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-04:
Using scripts, you can speed up the positioning of an element.
https://github.com/FreeCAD/FreeCAD/issues/14825

But of course the point you mentioned is not obvious.

yorikvanhavre, 2024-09-05:
Indeed the current way is a bit weak. Line https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/bimcommands/BimProfile.py#L119 should be changed with something more powerful, like:
```python
FreeCADGui.doCommand('p.Placement = FreeCAD.DraftWorkingPlane.get_placement()')
```

furgo16, 2024-09-05:
https://github.com/FreeCAD/FreeCAD/pull/16305 - set to draft pending further testing

PR #16305 (2024-09-10): Get working plane when picking a point
Fixes: #16284
commit 20065e64 (2024-09-10): BIM: Fixes profile tool positioning (#16305)

* [BIM] Get working plane when picking a point

Fixes: #16284

* Use the new way of getting the Draft Working Plane