FreeCAD/FreeCAD #16286: BIM: wrongly numerically increased profile name label

furgo16, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Switch to BIM workbench
2. Click on the BIM_Beam command
3. Select a category (e.g. "Eurotimber") and a preset (e.g. "22x100") on the Tasks dialog
4. Click on a start and end point to create the beam
5. Repeat steps 2 to 4 to create another beam
6. Expand both Beam objects in the document tree
7. Notice the underlying profile names: the first one is correct ("22x100"), whereas the second is not, as the name itself has been increased by one to "22x101"

If the reason for this change in names is to avoid duplicate labels, it would be more consistent to use the original label and append (not replace) an increasing numerical suffix ("000", "001", etc.). In the same way it's done for the parent Beam components in the tree.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38635 (Git) Snap 1125
Build type: Release
Branch: main
Hash: 7274dac185339a45020f1dc91f2b6c7478b2b0aa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.30
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-04:
22x100
22x100.001
22x100.002
22x100.003 
???

Roy-043, 2024-09-04:
The easiest solution is:
22x100_
22x100_001
22x100_002
etc.

yorikvanhavre, 2024-09-05:
@Roy-043 's solution is indeed very easy to implement... One just need to change line https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/Arch.py#L481 to add a "_" character at the end.

Anyone fancies doing it? ;)

furgo16, 2024-09-05:
Happy to: https://github.com/FreeCAD/FreeCAD/pull/16304

I've set it to Draft momentarily, as I need to test it properly during the course of the day.

yorikvanhavre, 2024-09-05:
Excellent! Thanks!!

> I've set it to Draft momentarily, as I need to test it properly during the course of the day.

Yes that's exactly what the draft status is for, so you can run the CI tests and any other test you want at ease... Removing the draft status is your way to notify "this is ready to merge".

PR #16304 (2024-09-09): [BIM] Separate profile name from label number
Fixes: #16286

This PR simply appends an underscore (`_`) at the end of the profile name's label. This means that the numerical sequence when adding multiple structures with a profile will be:

1. `$PROFILE`_ (e.g. 22x100_)
1. `$PROFILE`_ (e.g. 22x100_001)
1. `$PROFILE`_ (e.g. 22x100_002)

[image]

1. Notice that the first profile will only have an underscore suffix. The expectation here might be to either have no underscore, or (for more consistency) to start the sequence with the `_000` suffix. However, the assumption (to be corrected if wrong) i [... cut]
commit 17f74188 (2024-09-09): [BIM] Separate profile name from label number (#16304)

* [BIM] Separate profile name from label number

Fixes: #16286

* Make internal name generic
commit b7514bf8 (2024-09-09): [BIM] Separate profile name from label number (#16304)

* [BIM] Separate profile name from label number

Fixes: #16286

* Make internal name generic
closed 2024-09-09 by yorikvanhavre