FreeCAD/FreeCAD #16291: Core: Objects tree unfold whenever there is something broken.

leoheck, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most annoying behavior ever.

Why when something breaks somewhere, the whole tree starts to unfold? This is annoying as hell.
I cannot fix EVERYTHING at once. Them I have to keep closing the whole thing again and again.

Can you please, fix this?

If something is breaking, it will tell me on the icons of each object, the console view, and also the other extra notification area.

Please, KEEP the OBJECT TREE as the user wants! 
Do we have an option for that? IF YES, check if by default, PLEASE. 

`Std_Part`, `PartDesign_Body`, and `Std_Group` (and maybe others) should be as they are.

Please, if this is a must of behavior, give us a tool to find problematic issues. But keep the file tree as the user wants!

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-04:
@leoheck please check out https://github.com/FreeCAD/FreeCAD#reporting-issues
You've been reporting too many issues or duplicates which are not actionable and lack basic information. We always ask for the same and you should know by now.

leoheck, 2024-09-04:
@maxwxyz I am not reporting issues. I am busy working. But there are so many bugs and annoying things that I have to report. Sometimes I cannot find other related issues, but you do since you know the base of issues. Also, If there are missing information it is fine, just let me know and then I will fill the gaps when when I have time, later at the night or at dawn.

luzpaz, 2024-09-06:
Thanks for your efforts and engagement @leoheck. They really are appreciated. In the heat of the moment you are dealing with frustration and that's understandable. If you could notate that you will revisit ticket later with additional info would be helpful for the bug triage process. I think we can make an exception for leoheck, right @maxwxyz ? It is clear that they are digging deep into FC inner-workings and taking time to report issues.

leoheck, 2024-09-06:
This issue is complicated to reproduce. It needs things breaking on multiple modules. I don't have an example like that which I can also share right now. 

I will try to make an example, or reproduce it somehow. But it will take time, I think.

However the Bodies, Assemblies, Groups should not expand every time something fails. When it happens, the small list of objects turns into something really long to scroll back and forth everytime, specially when trying to fix things.

This happens since forever I think.

leoheck, 2024-09-06:
This is an hypothetical list of steps to reproduce:

- Create a big and complex project with multiple objects, Bodies, Parts, folders, links.. Make links to Bodies and reuse things.
- Fold everything (Groups, Bodies, Parts), specially on the root level of the objects tree.
- Make something fail, for instance on a Part Design body. 
- See the a bunch of things to unfold changing the current form of the objects tree.

leoheck, 2024-09-06:
There is something that I am noticing now.

Something appears on the Python Console when I click on an Object on the tree view.
However, when I fold/unfold something, nothing is shown on this console. 

Perhaps this is a default behavior of these things and something should be coded to prevent that.

leoheck, 2024-09-06:
Ah, this thing it is not called fold/unfold. It is called expand/collapse.
Found this to help me idenfity when this thing is triggered. https://wiki.freecad.org/Macro_ExpandTreeItem

leoheck, 2024-12-06:
This still happens.

leoheck, 2024-12-06:
When something breakes (let's say inside a Body). All the places where a link of this part exists, unfold, and the focus move to the end of the Object Tree (or to the least broken object). This behavior is disgusting and horrible to work with, specially on complex designs that have more objects than the space of the Objects tree. It is really hard to work having to scroll the focus all the way up on every single change.

leoheck, 2025-04-26:
Thanks @ElementW for detailing this. If something is breaking, please, print the object Name/Label to the console. This would help but not "scrolling to the item".

ElementW, 2025-04-27:
Printing the object name to the Report View already happens when the feature gets recomputed and is broken. Repeating their names in that view would be unnecessarily verbose.

Plus, I'd argue the Report View should never be necessary during normal (non-dev) FC use. UI work needs to be done to indicate a (sub)element of the tree has an error more clearly without being invasive like it is right now.
I don't use them but AFAIU:
* Fusion360 shows errored features in red in the timeline
  [image]
* Solidworks propagates the error up the tree with a matching downward red arrow symbol
  [image]
* Onshape seems to do the same
  [image]

Since FreeCAD's workflow has both linear (Part Design) and tree (Part & others) workflows, a solution like the latter two is probably the best fit.

leoheck, 2025-07-19:
This still happen on the latest weekly. It is annoying as hell.

spontarelliam, 2026-10-16:
@ElementW have you made a PR for your suggested change? I'd love to test it.

ElementW, 2026-10-16:
@spontarelliam If you mean the `scrollToItem(errItem);` removal: not yet.
If you mean the latter suggestions to align ourselves onto what Solidworks, Onshape etc do, no and I will not. This is a much larger discussion happening over at #32498's UI/UX discussion thread for both user interface and code design for error/warning/related feature reporting.

spontarelliam, 2026-10-16:
Oh I just mean scrollToItem. Please tag me when you post it. I also find this issue very annoying.