FreeCAD/FreeCAD #16292: FreeCAD exports "empty" STEP file for model, no errors in Check Geometry

theosib, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to export the object named "Cut" from the attached model, as a STEP file. I ran Check Geometry, and it reports no errors. And I can export this as an STL file with no problem. But if I export as a STEP file, both Bambu Studio and Prusa Slicer report that it has no geometry. And if I import it back into FreeCAD, the import is also blank.

No errors are reported in the console when exporting.

This also isn't the first time I've had to fall back to STL when FreeCAD would produce an invalid STEP file, BTW, so there's likely some kind of bug.

[smallpot.FCStd.zip](https://github.com/user-attachments/files/16874703/smallpot.FCStd.zip)
[smallpot-Cut.step.zip](https://github.com/user-attachments/files/16874707/smallpot-Cut.step.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-09-04:
It turns out that I didn't have enabled the option "Write out curves in parametric space of surface." It may be that my model is some kind of corner case that doesn't work with this option disabled.

luzpaz, 2024-09-05:
> It turns out that I didn't have enabled the option "Write out curves in parametric space of surface." 
>  It may be that my model is some kind of corner case that doesn't work with this option disabled.

Should we take this edge-case into consideration ?

coph, 2024-10-05:
I'm facing the same issue. I'm on a WIndows 11 computer. STEP file is empty while STL file looks good. It's a simple object. The "Write out curves in parametric space of surface." parameter is enabled.

d-den, 2024-11-27:
I'm having the same issue here since the update to the stable version 1.0.0.

tarman3, 2024-11-27:
Confirm.
I think the problem is with the object <ins>Sketch_On_Surface001</ins>

<details>
  <summary>FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51

```
</details>

FEA-eng, 2026-04-25:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260421 (Git shallow)
Build date: 2026/04/21 23:21:45
Build type: Release
Branch: (HEAD detached at 449725202)
Hash: 449725202d0ab581516aacad59d62cdce80db82a
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

Harrietofthesea, 2026-08-15:
I'm having the same issue. Step file exports perfectly via a python command, but will not export via the normal export function - results in a 2KB file. Stl also doesn't work. 
I'm on Mac, version 1.1.3