FreeCAD/FreeCAD #16293: BIM: Add option to exclude invisble objects from IFC export

Roy-043, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting Std_Parts containing Part Boolean objects to IFC, the invisible tool objects of the Boolean operations are included in the export. This is undesirable. To avoid this a new IFC export option should be added.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=89349

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-05:
Yes, makes sense.

maxwxyz, 2026-11-07:
This is implemented for the legacy IFC exporter in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/67c269ddb967f2857e77b92badf4912070909f0c

Exclude hidden objects filters hidden children, including nested groups. Containers and objects explicitly selected for export are kept.

furgo16, 2026-11-07:
@maxwxyz I would suggest doing these updates silently. That is, simply add the label, without a comment, given that:

- The issues already add a link to the commits automatically
- Everyone subscribed is flooded with new message notifications