FreeCAD/FreeCAD #16297: PartDesign Pad: To-First and To-Face seem not to be working again.

leoheck, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part Design Pad `To-Face` and `To-First` seems not to be working on this example.

I have this playground here [xxx.zip](https://github.com/user-attachments/files/16875572/xxx.zip)

The small bosses cylinders should grow until they touch the thin and wide circular rib.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
Confirmed. Saving the file in V0.21 and then reopening in V0.22 solves the issue. So it may not be PartDesign related?

FlachyJoe, 2024-10-07:
UpToFirst can't work because of the use of the profile center of gravity to find the cut face. As the wanted face is empty at this center it's not found.
UpToFace works on **1.1.0dev.38923**

luzpaz, 2025-11-28:
@FlachyJoe any chance this can get fixed before 1.1 release ?

FlachyJoe, 2025-11-29:
@luzpaz I have a try here : #25792 but it doesn't work as expected yet.

maxwxyz, 2026-05-14:
Is this still an issue?