FreeCAD/FreeCAD #16299: Some remaining duplicated extensions on export (FEM mesh formats)

FEA-eng, 2024-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As we know, #15838 introduced some regression that was then fixed by #15947. However, it seems that some export formats still exhibit this behavior. This is the case at least for FEM mesh formats.
1. Create Part Cube.
2. Go to FEM. Select the Cube and mesh it with Gmsh.
3. Select FEMMeshGmsh and File --> Export. Change the format from .dat to .unv and export.
4. Try exporting again. It shows Unnamed-FEMMeshGmsh.unv.dat this time.

This doesn't happen in 0.21.2. It might be the same with some other formats so I don't know if it's FEM-specific.

@mario

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-07:
@pieterhijma could you take a look at this since #15838 created this regression ? TIA

pieterhijma, 2024-09-08:
Yes, I will.

pieterhijma, 2024-09-08:
It is not FEM-specific. It has to do with file patterns that have more than one option, for example "Point formats (*.asc *.pcd *.ply)" or "FEM mesh Fenics (*.xml *.xdmf)". Choosing any of the extensions except for the first results in this situation.

PR #16374 (2024-09-10): Core: Fix duplicated extensions (see #16299)
This PR aims to resolve #16299, while taking into account https://github.com/FreeCAD/FreeCAD/issues/15908 and https://github.com/FreeCAD/FreeCAD/issues/15203 as well.

It simplifies the code and the regression happened because the regex was not clear to me initially and later also not to @chennes (I believe). I initially assumed the regex was meant to match all the suffixes in the description and probably @chennes took over this assumption because my code used it as such. However, I now understand that the regex was only meant to match the first ext [... cut]
closed 2024-09-10 by chennes