FreeCAD/FreeCAD #16310: If a sketch is attached to another boys face, this body can't be hidden

Panda631, 2024-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a simple body. In my case a box.
2. Create swcond body001 and a sketch inside (e.g. a circle) and pad it.
3. Disable the visibility of the first body.
4. Result: It is still visible.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Anything else?

[VisibleBody.zip](https://github.com/user-attachments/files/16893257/VisibleBody.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
The problem is different. The box is included in Body001. You can verify this by changing the Placement of Body.

I am surprised that it is allowed to attach a sketch like this.

Panda631, 2024-09-06:
> The problem is different. The box is included in Body001. You can verify this by changing the Placement of Body.
> 
> I am surprised that it is allowed to attach a sketch like this.

Apart from TNP problems, this should be a normal workflow. At least with v0.21 its not a problem.

Roy-043, 2024-09-06:
The same problem occurs in V0.21. Also there is a warning when you attach the sketch in V0.21:
Sketcher::SketchObject: Link(s) to object(s) 'Box' go out of the allowed scope 'Sketch'. Instead, the linked object(s) reside within 'Body Body'.

I would not call this a 'normal workflow'.

Panda631, 2024-09-07:
> The same problem occurs in V0.21. Also there is a warning when you attach the sketch in V0.21: Sketcher::SketchObject: Link(s) to object(s) 'Box' go out of the allowed scope 'Sketch'. Instead, the linked object(s) reside within 'Body Body'.
> 
> I would not call this a 'normal workflow'.

You are right, it happens also in v0.21. Sorry for that. I maybe overviewed the warning .

tarman3, 2024-12-07:
Bug or feature, but still relevant

<details>
  <summary>1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>