FreeCAD/FreeCAD #16316: SIGABRT - SketcherGui::SoZoomTranslation::initClass() (SoZoomTranslation.cpp:66)

huguesdpdn-aerospace, 2024-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Git pull the lastest version of FreeCAD with the RC1 commit and compile with debug mode

Instructions to reproduce:
```
git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD.git FreeCAD
cd FreeCAD
mkdir build
cmake -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Debug ../
-- Compiler: GNU, version: 13.2.0
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.28.3
-- Compiling with Qt 5
-- Checking for connection to GitHub...
-- GitHub connection established for FetchContent
-- Module support is disabled.
-- Version: 9.1.0
-- Build type: Debug
-- CXX_STANDARD: 17
-- Required features: cxx_variadic_templates
-- fmt was downloaded using FetchContent into /home/xxxxxxxx/FreeCAD/build/_deps/fmt-src
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter Development Development.Module Development.Embed 
-- PyCXX found:
--   Headers:  /home/xxxxxxxx/FreeCAD/src
--   Sources:  /home/xxxxxxxx/FreeCAD/src/CXX
--   Version:  7.1.9
-- -- Found OCE/OpenCASCADE version: 7.6.3
-- -- OCE/OpenCASCADE include directory: /usr/include/opencascade
-- -- OCE/OpenCASCADE shared libraries directory: /usr/lib/x86_64-linux-gnu
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.65") found components: filesystem program_options regex system thread date_time 
-- VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;IOMPIParallel;ParallelMPI;hdf5;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so  found components: OpenGL GLX 
-- Found HDF5: hdf5;/usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so;/usr/lib/x86_64-linux-gnu/libcrypto.so;/usr/lib/x86_64-linux-gnu/libcurl.so;/usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/libsz.so;/usr/lib/x86_64-linux-gnu/libz.so;/usr/lib/x86_64-linux-gnu/libdl.a;/usr/lib/x86_64-linux-gnu/libm.so (found version "1.10.10") found components: C HL 
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0") 
-- Check for medfile (libmed and libmedc) ...
-- We guess that libmed was built using hdf5-serial version
-- Set up to compile with Qt 5.15.13
-- Checking Pivy version by importing it in a Python program...
-- Found Pivy 0.6.9
-- Shiboken2Config: Using default python: .cpython-312-x86_64-linux-gnu
-- SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: '/usr/include/python3.12'
-- SHIBOKEN_PYTHON_LIBRARIES computed to value: ''
-- libshiboken built for RelWithDebInfo
-- PYTHON_CONFIG_SUFFIX: .cpython-312-x86_64-linux-gnu
-- libshiboken built for RelWithDebInfo
-- PySide 5.15.13 Python module found at /usr/lib/python3/dist-packages/PySide2.

-- Found Matplotlib: /usr/lib/python3/dist-packages/matplotlib (found version "3.6.3")
-- Platform is 64-bit, set -D_OCC64
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
CMake Warning (dev) at src/3rdParty/libE57Format/cmake/Modules/Tags.cmake:31 (exec_program):
  Policy CMP0153 is not set: The exec_program command should not be called.
[... 228 lines cut ...]
Will try to investigate a bit from my side.

### Full version info

```shell
Cannot open FreeCAD dialog version since the program crashes before even opening the main window. Here is the last GIT commit:

commit 0026a35886cf7b62d5fa5861ba1e52f7434f84e2 (HEAD -> main, origin/main, origin/HEAD)
Author: Chris Hennes <chennes@gmail.com>
Date:   Wed Sep 4 13:22:02 2024 -0500

    Core: Update version number to 1.0.0RC1

My OS information:
$> uname --all
Linux DELLINSPIRON16 6.8.0-41-generic #41-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug  2 20:41:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-09-05:
OK so I dig a lot this evening.

## Investigation
I found out that the specified crashing line, aka [SoZoomTranslation.cpp line 66](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/SoZoomTranslation.cpp#L66) use a library called OpenInventor where the `SO_ENABLE` macro is defined and used.

`SO_ENABLE` : [Documentation](https://developer9.openinventor.com/UserGuides/9.9/Inventor_Toolmaker/Key_Concepts/Actions_State_and_Elements.html)
`SO_ENABLE` : [Source code #1 - Line 814](https://developer.openinventor.com/refmans/10.12/RefManCpp/_so_action_8h_source.html)
`SO_ENABLE` : [Source code #2 - Line 445](https://mevislabdownloads.mevis.de/docs/current/MeVis/ThirdParty/Documentation/Publish/OpenInventorReference/SoAction_8h_source.html#l00445)

The issue is that, even if there are 2 differents versions above, the include header installed on my system is different from both links. I have a third version - The following installed in my `/usr/local/include/Inventor/actions/SoAction.h` :

```
#define SO_ENABLE(action, element) \
  do { \
    assert(!element::getClassTypeId().isBad()); \
    action::enableElement(element::getClassTypeId(), \
                          element::getClassStackIndex()); \
  } WHILE_0
```
and in this version, there is an `assert` call, not in the one from the OpenInventor source code from the above links.

Between, if I test the value of `element::getClassTypeId().isBad()`, the value is True.
Also, `element::getClassTypeId().getName()` is "BadType".
However, if I manually perform the call `action::enableElement(element::getClassTypeId(), element::getClassStackIndex())` instead of the `SO_ENABLE` macro call, it works. No error triggered.

So I checked with `apt-file find SoAction.h` and it gives me 2 results (I added the version):
```
inventor-dev   2.1.5-10+dfsg-2.1build2   /usr/include/Inventor/actions/SoAction.h
libcoin-dev    4.0.2+ds-1.1ubuntu2       /usr/include/Inventor/actions/SoAction.h
```

Let have a look in the package manager:
- in `inventor-dev` ([from the download package inventor-dev_2.1.5-10-21build1_amd64.deb](http://ftp.ubuntu.com/ubuntu/pool/universe/i/inventor/inventor-dev_2.1.5-10-21build1_amd64.deb)), there is no `assert` call
- in `libcoin-dev` ([from the download package libcoin-dev 4.0.2+ds-1.1ubuntu2_amd64.deb](http://launchpadlibrarian.net/722372892/libcoin-dev_4.0.2+ds-1.1ubuntu2_amd64.deb)), there is my current `/usr/local/include/Inventor/actions/SoAction.h` header file.

## Now what ?

My first question would be: is my `SoAction.h` a very old file ?
- if yes, why Ubuntu has a very old package in their repo ?
- if no, then we need to handle this in FreeCAD

My second question would be: should the package `inventor-dev` be installed over the `libcoin-dev` package ?
- if yes, then the [compile guide on Linux](https://wiki.freecad.org/Compile_on_Linux) is incorrect since it requires to install `libcoin-dev` and not `inventor-dev`
- if no, all good

My third and last question would be: should we implement, in the FreeCAD source code in file [SoZoomTranslation.cpp at line 66](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/SoZoomTranslation.cpp#L66), a check about the COIN version used and if recent, change the code to avoid this `assert()` crash.

luzpaz, 2024-09-06:
Really great sleuthing. Looks like you uncovered a [coin3d](https://github.com/coin3d/coin) bug ?

huguesdpdn-aerospace, 2024-09-06:
I am not sure this is a coind3d bug, otherwise many other people would already reported this issue to Coin3D. I think this `assert` call is legit. I (unfortunately) think this error is more on FreeCAD side :(

xtemp09, 2024-09-07:
This is so interesting. Great investigation. 👍 However, some questions remain.

- You reported a crash, but you haven't provided the backtrace. Could you upload it?

- How can you possibly encounter `assert` if _coin_ is compiled with `NDEBUG`? How do you compile _coin_? Its compilation is quite tricky.

huguesdpdn-aerospace, 2024-09-07:
@xtemp09 : 
 
- The backtrace has been uploaded in the Valgrind log file in my first comment. But you can consult it just below :)
Full vallgrind traces in [FreeCAD_v1.0.0.RC1_Valgrind_SIGABRT.log](FreeCAD_v1.0.0.RC1_Valgrind_SIGABRT.log) (click on the link to download the full log file). Our issue only starts at line 4523 . Here is the extract:
```
==47291== Process terminating with default action of signal 6 (SIGABRT)
==47291==    at 0xBDC3B1C: __pthread_kill_implementation (pthread_kill.c:44)
==47291==    by 0xBDC3B1C: __pthread_kill_internal (pthread_kill.c:78)
==47291==    by 0xBDC3B1C: pthread_kill@@GLIBC_2.34 (pthread_kill.c:89)
==47291==    by 0xBD6A26D: raise (raise.c:26)
==47291==    by 0xBD4D8FE: abort (abort.c:79)
==47291==    by 0xBD4D81A: __assert_fail_base.cold (assert.c:94)
==47291==    by 0xBD60506: __assert_fail (assert.c:103)
==47291==    by 0x3C652CD0: SketcherGui::SoZoomTranslation::initClass() (SoZoomTranslation.cpp:66)
==47291==    by 0x3C4371DD: PyInit_SketcherGui (AppSketcherGui.cpp:136)
==47291==    by 0xA09C41E: ??? (in /usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0)
==47291==    by 0xA0993A9: ??? (in /usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0)
==47291==    by 0x9FA2764: ??? (in /usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0)
==47291==    by 0x9EDFDEB: _PyEval_EvalFrameDefault (in /usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0)
==47291==    by 0x9F48D6D: ??? (in /usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0)
==47291==
==47291== HEAP SUMMARY:
==47291==     in use at exit: 64,542,148 bytes in 142,960 blocks
==47291==   total heap usage: 836,621 allocs, 693,661 frees, 545,753,153 bytes allocated
```
Sorry, I don't have a better trace than this one and nothing else to provide. 

- I didn't compiled `coin` by myself but I have `/usr/local/lib/libCoin.so.80 -> libCoin.so.80.0.2` installed. The `libcoin-dev` package has been installed from my officialUbuntu/Debian package manager, as requested in [the compile on Linux wiki](https://wiki.freecad.org/Compile_on_Linux) ; this is why it has the same version as the one use today in PR hooks (aka version `4.0.2`). However, `libcoin-dev` package, in Debian/Ubuntu, includes headers from `OpenInventor`. So Coin itself is all good but the headers installed with the `libcoin-dev` are ... incorrect ?

- How can you possibly encounter assert if coin is compiled with `NDEBUG`?
Because the `assert` is in a header file (was it compiled with `NDEBUG`, I don't know) in Coin BUT this same header file with the `assert` call is included in FreeCAD and then compiled (in Sketcher first and the pretty much everywhere else afterward)

- The real question is why in `libcoin-dev` package there is the `assert` call ?
I don't know, but the file is not even part of `Coin` itself but part of `OpenInventor` included from Coin.
The source of the `libcoin-dev` package can be downloaded here: [libcoin-dev_4.0.2+ds-1.1ubuntu2_amd64.deb](http://launchpadlibrarian.net/722372892/libcoin-dev_4.0.2+ds-1.1ubuntu2_amd64.deb) - You can check it by reading file, like:
```
wget http://launchpadlibrarian.net/722372892/libcoin-dev_4.0.2+ds-1.1ubuntu2_amd64.deb
ar vx libcoin-dev_4.0.2+ds-1.1ubuntu2_amd64.deb
tar xvf data.tar.zst
cat ./usr/include/Inventor/actions/SoAction.h
```

Hope my explanations help

huguesdpdn-aerospace, 2024-09-07:
UPDATE

@xtemp09 , you was a bit right. So I downloaded the official [Coind3D/coin](https://github.com/coin3d/coin) repo and then:
- `Debug` mode => `SoAction.h` has the `assert` call, and it is triggering the `SIGABRT`
```
==36124== Process terminating with default action of signal 6 (SIGABRT)
==36124==    at 0xBDC2B1C: __pthread_kill_implementation (pthread_kill.c:44)
==36124==    by 0xBDC2B1C: __pthread_kill_internal (pthread_kill.c:78)
==36124==    by 0xBDC2B1C: pthread_kill@@GLIBC_2.34 (pthread_kill.c:89)
==36124==    by 0xBD6926D: raise (raise.c:26)
==36124==    by 0xBD4C8FE: abort (abort.c:79)
==36124==    by 0xBD4C81A: __assert_fail_base.cold (assert.c:94)
==36124==    by 0xBD5F506: __assert_fail (assert.c:103)
==36124==    by 0xC63CE29: SoType::isDerivedFrom(SoType) const (SoType.cpp:739)
==36124==    by 0xC612DDB: SoBase::isOfType(SoType) const (SoBase.cpp:633)
==36124==    by 0xC591010: SoGlobalField* coin_internal_safe_cast<SoGlobalField*, SoBase>(SoBase*) (SbBasicP.h:86)
==36124==    by 0xC590FE3: SoGlobalField* coin_safe_cast<SoGlobalField*>(SoBase*) (SbBasicP.h:97)
==36124==    by 0xC590F8E: SoGlobalField* coin_internal_assert_cast<SoGlobalField*, SoBase>(SoBase*) (SbBasicP.h:124)
==36124==    by 0xC590F70: SoGlobalField* coin_assert_cast<SoGlobalField*>(SoBase*) (SbBasicP.h:138)
==36124==    by 0xC5903EF: SoGlobalField::getGlobalFieldContainer(SbName const&) (SoGlobalField.cpp:216)
==36124==
==36124== HEAP SUMMARY:
==36124==     in use at exit: 57,167,861 bytes in 121,279 blocks
==36124==   total heap usage: 611,859 allocs, 490,580 frees, 308,185,671 bytes allocated
```

- `Release` mode =>`SoAction.h` has the `assert` call but the `NDEBUG` macro is enabled, so no immediate `SIGABRT`. We have a first `Jump to the invalid address` and again the `SIGABRT`.
```
==44334== Jump to the invalid address stated on the next line
==44334==    at 0x0: ???
==44334==    by 0xC247487: SoAction::traverse(SoNode*) (in /usr/local/lib/libCoin.so.80.0.3)
==44334==    by 0xC257E55: SoSearchAction::beginTraversal(SoNode*) (in /usr/local/lib/libCoin.so.80.0.3)
==44334==    by 0xC2480E1: SoAction::apply(SoNode*) (in /usr/local/lib/libCoin.so.80.0.3)
==44334==    by 0x6276A07: SIM::Coin3D::Quarter::QuarterWidgetP::searchForCamera(SoNode*) (QuarterWidgetP.cpp:118)
==44334==    by 0x6272FA4: SIM::Coin3D::Quarter::QuarterWidget::setSceneGraph(SoNode*) (QuarterWidget.cpp:635)
==44334==    by 0x627E6C8: SIM::Coin3D::Quarter::SoQTQuarterAdaptor::setSceneGraph(SoNode*) (SoQTQuarterAdaptor.h:109)
==44334==    by 0x62D251D: Gui::View3DInventorViewer::setSceneGraph(SoNode*) (View3DInventorViewer.cpp:1520)
==44334==    by 0x62CC6E2: Gui::View3DInventorViewer::init() (View3DInventorViewer.cpp:483)
==44334==    by 0x62CBF62: Gui::View3DInventorViewer::View3DInventorViewer(QSurfaceFormat const&, QWidget*, QOpenGLWidget const*) (View3DInventorViewer.cpp:397)
==44334==    by 0x62C179B: Gui::View3DInventor::View3DInventor(Gui::Document*, QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) (View3DInventor.cpp:114)
==44334==    by 0x5C4F6CD: Gui::Document::createView(Base::Type const&) (Document.cpp:1783)
==44334==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==44334==
==44334==
==44334== Process terminating with default action of signal 6 (SIGABRT)
==44334==    at 0xBDC2B1C: __pthread_kill_implementation (pthread_kill.c:44)
==44334==    by 0xBDC2B1C: __pthread_kill_internal (pthread_kill.c:78)
==44334==    by 0xBDC2B1C: pthread_kill@@GLIBC_2.34 (pthread_kill.c:89)
==44334==    by 0xBD6926D: raise (raise.c:26)
==44334==    by 0xBD4C8FE: abort (abort.c:79)
==44334==    by 0x93A7BC0: segmentation_fault_handler(int) (Application.cpp:1837)
==44334==    by 0xBD6931F: ??? (in /usr/lib/x86_64-linux-gnu/libc.so.6)
==44334==
==44334== HEAP SUMMARY:
==44334==     in use at exit: 78,573,456 bytes in 230,108 blocks
==44334==   total heap usage: 5,604,322 allocs, 5,374,214 frees, 1,022,931,199 bytes allocated
```

So even building the latest `libCoin` in `Release` or `Debug`, we get our `SIGABRT` and this is an issue in FreeCAD, not in Coin neither in `OpenInventor` as I thought so since the beginning

Will try to dig deeper when time permits

xtemp09, 2024-09-08:
> some UNIX testers : test if the issue is reproducible with a fresh Debian/Ubuntu install

I cannot reproduce it on _KUbuntu 24.04.1 LTS (Noble Numbat)_.

----

I noticed that the back trace begins with

> ==47291==    by 0x3C4371DD: PyInit_SketcherGui (AppSketcherGui.cpp:136)

Could you compile _FreeCAD_ with _pybind11_ on?

benj5378, 2024-10-06:
For some reason, this is an issue on my Desktop but not my laptop. Wondering if my laptop works due to EGL build

benj5378, 2024-10-06:
I fixed this by `sudo apt remove python3-pivy`. Idk if it's of any importance @huguesdpdn-aerospace

benj5378, 2024-10-07:
> My first question would be: is my SoAction.h a very old file ? Or in other words, is my libcoin library too old ?

No, I get a similar error using a new build from Coin on GitHub. I don't believe it's due to a new Coin version either. There are multiple examples of people running Coin 4.0.3 and have builded FreeCAD themselves without issues.

Some peolpe having similar issue in https://github.com/coin3d/coin/pull/531

galou, 2024-10-08:
The changes in the Coin library didn't help avoiding a crash. What did though is the [cited PR](https://github.com/FreeCAD/FreeCAD/pull/17073). I just tested for a short time (with the `eglDustOff2` branch, 9e7d403d24ac, and QT_QPA_PLATFORM=wayland), the main window opens, a can create a new document and a sketch with constructions lines. I had a crash though:
```
[destroyed object]: error 3: xdg_surface has never been configured
The Wayland connection experienced a fatal error: Protocol error
```

benj5378, 2024-10-18:
For some reason, this is still an issue on my Desktop even after the PR. It wasn't an issue before. Reverting the PR doesn't fix it. @luzpaz, can you reopen?

galou, 2024-11-20:
I managed to launch FreeCAD on Pop!_OS 24.04 (based on Ubuntu 24.04) and it seems to work  by using the workaround suggested by FreeCAD when launch with `QT_QPA_PLATFORM=wayland`, actually suggested by `gl.cpp`:
```
freecad-daily: ./src/glue/gl.cpp:2234: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
```

Launching FreeCAD with `COIN_GL_NO_CURRENT_CONTEXT_CHECK=1 QT_QPA_PLATFORM=wayland freecad-daily` works but there are some warnings:
```
FreeCAD 1.1.0, Libs: 1.1.0devR39241 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

QSocketNotifier: Can only be used with threads started with QThread
QOpenGLFunctions created with non-current context
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
This plugin supports grabbing the mouse only for popup windows
```

The last warning appeared when I was moving toolbars around.

OS: Pop!_OS 24.04 LTS (wayland)
Architecture: x86_64
Version: 1.1.0dev.39241 (Git)
Build type: Release
Branch: main
Hash: cdb92768fe4f33d900664415c613f9eced5b8927
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

PR #17073 (2024-10-14): [Gui] Fix isBad asserts by removing old monkey patch
Multiple places in code, asserts for bad types would trigger SIGABRT. This is not just limited to #16316 with SoZoomTranslation, but multiple more places, not described in issue, but I experienced myself. Fixes #16316
commit d4173be1 (2024-10-14): [Gui] Fix isBad asserts, by removing old monkey patch

Multiple places in code, asserts for bad types would trigger SIGABRT. This is not just limited to #16316 with SoZoomTranslation, but multiple more places, not described in issue, but I experienced myself. Fixes #16316
commit b16bd50f (2024-10-14): [Gui] Fix isBad asserts, by removing old monkey patch

Multiple places in code, asserts for bad types would trigger SIGABRT. This is not just limited to #16316 with SoZoomTranslation, but multiple more places, not described in issue, but I experienced myself. Fixes #16316