FreeCAD/FreeCAD #16319: Part.BSplineCurve interpolate(points) does not accept a list of App.Vector for points

edwilliams16, 2024-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

```help(bsp.approximate)
Help on built-in function approximate:

approximate(...) method of Part.BSplineCurve instance
    Replaces this B-Spline curve by approximating a set of points.
    The function accepts keywords as arguments.
    
    approximate(Points = list_of_points)
    
```
Here a list_of_points requires a list of tuples of floats.

whereas

```help(bsp.interpolate)
Help on built-in function interpolate:

interpolate(...) method of Part.BSplineCurve instance
    Replaces this B-Spline curve by interpolating a set of points.
    The function accepts keywords as arguments.
    
    interpolate(Points = list_of_points)
```

Here list_of_points also accepts a list of App.Vectors.   
It would be desirable to make these consistent, by allowing approximate() to accept a list of vectors. (and/or make the tooltips more explicit.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.10.8
  * DynamicData 2.62.0
  * OpenTheme 2024.8.17
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.4.20
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.45
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
AFAICT both approximate and interpolate accepts a list of vectors:
```python
import Part

pt1 = App.Vector()
pt2 = App.Vector(1, 0, 0)
pt3  =App.Vector(1, 1, 0)

bs1 = Part.BSplineCurve()
bs1.approximate([pt1, pt2, pt3])
Part.show(bs1.toShape())

bs2 = Part.BSplineCurve()
bs2.interpolate([pt1, pt3, pt2])
Part.show(bs2.toShape())
```

edwilliams16, 2024-09-06:
This definitely was not working - but now it is!  I must have found a way to corrupt the Python Console in my testing.

closed 2024-09-06 by luzpaz