FreeCAD/FreeCAD #16320: Model that exports fine to STL but not STEP

theosib, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm attaching another model that doesn't export properly to STEP. When I import it into a slicer, it has holes in it in strange places. I tried with and without the "write out curves in parametric space of surface" option, but both had problems.

If I export it as STL, t's fine.
[knurl-PartCut001.step.zip](https://github.com/user-attachments/files/16901465/knurl-PartCut001.step.zip)
[knurl.FCStd.zip](https://github.com/user-attachments/files/16901466/knurl.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
Are you sure this is a correct model?
[image]

theosib, 2024-09-06:
I made a reflected version of this curve but didn't use it. Also, I have "Export invisible objects" off, so this extra hidden element should not get exported. It's also not showing up in the step. Instead, I'm getting weird gaps in the inner wall. 
[image]

Roy-043, 2024-09-06:
I have not made  any objects visible. Please check the file you have provided. Also please explain what you export, and what your STEP export settings are.

theosib, 2024-09-06:
I tried importing the STEP file into a few more things, and it seems like only BambuStudio is having trouble with this. I imported it into PrusaSlicer and FreeCAD, and both imported it fine. So I'm going to file this as a bug on BambuStudio. It's possible that FreeCAD is doing something slightly weird, but I'm guessing probably not since the other imports work fine.

-------

I'm exporting the "Part" object, which is an Std_Part. This is the same object I export when exporting as STL, an that works fine. I'm attaching my project file again, along with the STEP file.

[knurl-Part.step.zip](https://github.com/user-attachments/files/16909370/knurl-Part.step.zip)
[knurl.FCStd.zip](https://github.com/user-attachments/files/16909372/knurl.FCStd.zip)

[image]

theosib, 2024-09-06:
https://github.com/bambulab/BambuStudio/issues/4761

PCS-Rimage, 2024-10-06:
Hello,

I discovered a similar issue using FC 1.0 RC1 and RC2.  I found that if I have some non-visible objects in a part then export as step fails with "Access violation" message.  The same part exports to stl without error and, as I would expect, does not include the non-visible objects.  If I make all objects visible in the part then step export works too.

luzpaz, 2025-07-19:
Is this still an issue ?

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```

mnesarco, 2026-11-13:
I have this problem with a much simple model.

# FreeCAD:

[image]

# STL:

[image]

# STEP:

[image]

# Document

[document.FCStd.zip](https://github.com/user-attachments/files/34818838/document.FCStd.zip)