FreeCAD/FreeCAD #16321: Sketcher - unable to create Ellipse by Center if a point is an Axis

arena-, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unable to create an Ellipse by Center if one of the subsequent selection points meets an Axis. 

To recreate:
New File->New Body->New Sketch->XY Plane
Select 'Ellipse by Center'
Click on Origin
Click first point on the X axis
Move the mouse pointer up and click the second point anywhere else, so it looks like a normal ellipse

- Sketcher will show just a straight line with a tangent icon over the Origin

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.5.24
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-30:
@amreo feel free to subscribe to this ticket in order to get updates

luzpaz, 2024-09-30:
@AjinkyaDahale could you weigh-in on this at your convenience ? TIA!

AjinkyaDahale, 2024-10-02:
Sounds like a regression. Could someone confirm?

AjinkyaDahale, 2024-10-02:
The problem is that what should be a point on object constraint on the axis is instead applied as a tangent. If you delete that and apply a length constraint on the other axis (which is made degenerate), we get an ellipse.

amreo, 2024-10-02:
> Sounds like a regression. Could someone confirm?

It seems to work correcly on 0.21.2 so I think this bug is a regression

AjinkyaDahale, 2024-10-02:
> It seems to work correcly on 0.21.2 so I think this bug is a regression

@amreo could you send a screenshot or file of what it looks like in 0.21.2?

amreo, 2024-10-02:
> > It seems to work correcly on 0.21.2 so I think this bug is a regression
> 
> @amreo could you send a screenshot or file of what it looks like in 0.21.2?

[image]
(note: the language is Italian)

```
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.5.25
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.9
  * boltsfc 2022.11.5
  * ThreadProfile 1.92.0
  * sheetmetal 0.4.23
  * lattice2 1.0.0
  * Curves 0.6.42
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.65
```

AjinkyaDahale, 2024-10-03:
The entire game is within `DrawSketchHandler::seekAutoConstraint` and `seekAndRenderAutoConstraint`. If we click on an existing curve while a point on the new curve is being decided (a generic point, nothing specific like an end point), a tangent constraint between the two curves is created. This should either not exist or be a perpendicular or other constraint (or no constraint at all). The signatures are a little curious, there is no documentation.
```
    int seekAutoConstraint(std::vector<AutoConstraint>& suggestedConstraints,
                           const Base::Vector2d& Pos,
                           const Base::Vector2d& Dir,
                           AutoConstraint::TargetType type = AutoConstraint::VERTEX);

    int seekAndRenderAutoConstraint(std::vector<AutoConstraint>& suggestedConstraints,
                                    const Base::Vector2d& Pos,
                                    const Base::Vector2d& Dir,
                                    AutoConstraint::TargetType type = AutoConstraint::VERTEX);
```

Is `Dir` a normal or tangent direction? Looking at the code, a similar variable `hitShapeDir` is used to get a tangent direction. But these lines are curious:

https://github.com/FreeCAD/FreeCAD/blob/ffa1189634fd09bbcfe40639db2cc90b24ee4797/src/Mod/Sketcher/Gui/DrawSketchHandler.cpp#L524-L528

This implies that the tangency is only applied if the angle between `Dir` and `hitShapeDir` is _more_ than about 6 degrees (`cosangle < 0.885`), which sounds counterintuitive, unless `Dir` is for normal, and `hitShapeDir` is for tangent direction.

A search into the uses of these methods suggests that only a zero vector is passed for `Dir`, except for a few cases, which suggest it is a normal. @PaddleStroke could you weigh in, since these lines were last edited by you?

So currently I see a couple of solutions possible:
1. Standardize the use of `Dir` (in doc string) and `hitShapeDir` (in code usage) to normal, and actually use these values. Possibly also apply perpendicular instead, but this may be considered a feature. Also, do not create any constraint if the two curves do not support direct curve to curve tangency (without a common point). This would be a stop-gap solution for 1.0.
2. In cases where we could instead use 2 point-on-objects (or tangent-via-point) with an extra point added, do that. This would be a longer term solution: it requires a queue of geometries to be created at completion of command. As a sidenote, this could also be used to address #12473.

commit 596fa285 (2024-10-04): [Sketcher] Apply tangent autoconstraint only when `Dir` given

This relates to `seekAutoConstraint` and `seekAndRenderAutoConstraint`.

Fixes #16321.
commit 7acf6274 (2024-10-04): [Sketcher] Apply tangent autoconstraint only when `Dir` given

This relates to `seekAutoConstraint` and `seekAndRenderAutoConstraint`.

Fixes #16321.
closed 2024-10-04 by chennes