FreeCAD/FreeCAD #16323: PartDesign Chamfer/Fillet: Segfault!

leoheck, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A simple Chamfer/Fillet is breaking the tool.
My file [chamfer_fillet_issue.zip](https://github.com/user-attachments/files/16902638/chamfer_fillet_issue.zip)

For instance, when selecting this this edge here: 
(I cannot copy the edges to text =|) 

[image]

Straight edge break the tool:
[image]

However if I select the an round edge, like this, it works:
[image]

Round edge does not break the tool.
[image]

And the operation also works.. (it just has to be a bit less then the default 1 mm)

[image]

```diff
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x715cfca45320]
#1  0x715cd8d57784 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) from /tmp/.mount_FreeCAplOhzD/usr/lib/./libTKBRep.so.7.7+0x164
#2  0x715cd8d5785f in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) from /tmp/.mount_FreeCAplOhzD/usr/lib/./libTKBRep.so.7.7+0xaf
#3  0x715cd8d659ec in BRepAdaptor_Curve2d::Initialize(TopoDS_Edge const&, TopoDS_Face const&) from /tmp/.mount_FreeCAplOhzD/usr/lib/./libTKBRep.so.7.7+0xdc
#4  0x715cb2d33d80 in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_Curve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_Surface>&, opencascade::handle<BRepAdaptor_Curve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const from /tmp/.mount_FreeCAplOhzD/usr/lib/././libTKFillet.so.7.7+0xd50
#5  0x715cb2d39204 in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /tmp/.mount_FreeCAplOhzD/usr/lib/././libTKFillet.so.7.7+0x684
#6  0x715cb2d3b9e1 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /tmp/.mount_FreeCAplOhzD/usr/lib/././libTKFillet.so.7.7+0x101
#7  0x715cb2d4115a in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /tmp/.mount_FreeCAplOhzD/usr/lib/././libTKFillet.so.7.7+0x6a
#8  0x715cb2d0515d in ChFi3d_Builder::Compute() from /tmp/.mount_FreeCAplOhzD/usr/lib/././libTKFillet.so.7.7+0x42d
#9  0x715cb2e93b2e in BRepFilletAPI_MakeChamfer::Build(Message_ProgressRange const&) from /tmp/.mount_FreeCAplOhzD/usr/lib/././libTKFillet.so.7.7+0x2e
#10  0x715cc083b1ed in BRepBuilderAPI_MakeShape::Shape() from /tmp/.mount_FreeCAplOhzD/usr/lib/./libTKTopAlgo.so.7.7+0x8d
#11  0x715cc0ef1a1f in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /tmp/.mount_FreeCAplOhzD/usr/lib/./Part.so+0x7f
#12  0x715cc0ef1bc3 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /tmp/.mount_FreeCAplOhzD/usr/lib/./Part.so+0x83
#13  0x715cc0ef38b0 in Part::TopoShape::makeElementChamfer(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, Part::ChamferType, double, double, char const*, Part::Flip) from /tmp/.mount_FreeCAplOhzD/usr/lib/./Part.so+0x3b0
#14  0x715c9759be6a in PartDesign::Chamfer::execute() from /tmp/.mount_FreeCAplOhzD/usr/lib/_PartDesign.so+0x2aa
#15  0x715cfec624a2 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADApp.so+0xc2
#16  0x715cc0ca52ec in Part::Feature::recompute() from /tmp/.mount_FreeCAplOhzD/usr/lib/./Part.so+0xc
#17  0x715cfec3c50c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADApp.so+0x32c
#18  0x715cfec48af5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADApp.so+0x625
#19  0x715cfecc774e in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADApp.so+0x2ee
#20  0x715cfecc7864 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADApp.so+0x24
#21  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x715cff406068]
#22  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x715cff3e5b5c]
#23  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x715cff3f1532]
#24  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x715cff4abd1d]
#25  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x715cff4ab794]
#26  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x715cff4cabf3]
#27  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x715cff4c669a]
#28  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x715cff4ba9c9]
#29  0x715cfe73d9fc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADBase.so+0x6c
#30  0x715cffd89b64 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADGui.so+0x94
#31  0x715cffd89cf3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADGui.so+0xe3
#32  0x715cffd89e14 in Gui::Command::updateActive() from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADGui.so+0x44
#33  0x715c7c8fc566 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /tmp/.mount_FreeCAplOhzD/usr/lib/PartDesignGui.so+0x1d6
#34  0x715c7c900600 in finishDressupFeature(Gui::Command const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Part::Feature*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) from /tmp/.mount_FreeCAplOhzD/usr/lib/PartDesignGui.so+0x620
#35  0x715c7c90e81e in makeChamferOrFillet(Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /tmp/.mount_FreeCAplOhzD/usr/lib/PartDesignGui.so+0xde
#36  0x715c7c90eb49 in CmdPartDesignChamfer::activated(int) from /tmp/.mount_FreeCAplOhzD/usr/lib/PartDesignGui.so+0x49
#37  0x715cffd93ce8 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADGui.so+0x2a8
#38  0x715cffd940a4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libFreeCADGui.so+0x134
#39  /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x715cfd0eb6f2]
#40  0x715cfdf6a462 in QAction::triggered(bool) from /tmp/.mount_FreeCAplOhzD/usr/bin/../lib/libQt5Widgets.so.5+0x32
[... 26 lines cut ...]
#67  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x715cfca2a1ca]
#68  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x715cfca2a28b]
#69  /tmp/.mount_FreeCAplOhzD/usr/bin/freecad(+0x79a9) [0x5aec159c49a9]
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
On Windows there is an Access violation error. If I then reduce the Chamfer size from the default value of 1.0 to 0.5 the Chamfer succeeds.

leoheck, 2024-09-06:
Good, this happens sometimes, however, for me it is breaking before of showing the Chamfer/Fillet settings.

jstruebel, 2024-11-30:
I am having a similar issue that might be related to this. Let me know if it's different and I'll open a new issue.

Sample File: [Fillet_Crash_example.zip](https://github.com/user-attachments/files/17964211/Fillet_Crash_example.zip)

When I try adding Edge48 to Fillet003 (or any other edge that is on the top of that pillar) I get a Segmentation Fault when I click the OK button to complete the operation. 

[image]

Changing from select mode to preview mode shows the edge added to the fillet as desired.

[image]

Error produced when clicking OK button following operations above.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f14b2c5b050]
#1  0x7f14b4eb4854 in Data::ElementNameComparator::operator()(Data::MappedName const&, Data::MappedName const&) const from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADApp.so+0x24
#2  0x7f147cb098ac in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Part::NameInfo>, std::_Select1st<std::pair<Data::MappedName const, Part::NameInfo> >, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Part::NameInfo> > >::_M_get_insert_hint_unique_pos(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Part::NameInfo> >, Data::MappedName const&) from /tmp/.mount_FreeCAlilPPK/usr/lib/./Part.so+0x5c
#3  0x7f147cb17124 in Part::TopoShape::makeShapeWithElementMap(TopoDS_Shape const&, Part::TopoShape::Mapper const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /tmp/.mount_FreeCAlilPPK/usr/lib/./Part.so+0x4ff4
#4  0x7f147cb1892e in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /tmp/.mount_FreeCAlilPPK/usr/lib/./Part.so+0x11e
#5  0x7f147cb18a33 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /tmp/.mount_FreeCAlilPPK/usr/lib/./Part.so+0x83
#6  0x7f147cb1a01f in Part::TopoShape::makeElementFillet(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, double, double, char const*) from /tmp/.mount_FreeCAlilPPK/usr/lib/./Part.so+0x10f
#7  0x7f1465d9922a in PartDesign::Fillet::execute() from /tmp/.mount_FreeCAlilPPK/usr/lib/_PartDesign.so+0x21a
#8  0x7f14b4c741f2 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADApp.so+0xc2
#9  0x7f147c8cb47c in Part::Feature::recompute() from /tmp/.mount_FreeCAlilPPK/usr/lib/./Part.so+0xc
#10  0x7f14b4c4914c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADApp.so+0x32c
#11  0x7f14b4c587ed in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADApp.so+0x5fd
#12  0x7f14b4cd9c31 in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADApp.so+0x301
#13  0x7f14b4cd9d54 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADApp.so+0x24
#14  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f14b5406068]
#15  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f14b53e5b5c]
#16  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f14b53f1532]
#17  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f14b54abd1d]
#18  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f14b54ab794]
#19  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f14b54cabf3]
#20  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f14b54c669a]
#21  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f14b54ba9c9]
#22  0x7f14b474641c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADBase.so+0x6c
#23  0x7f14b5db6ba4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADGui.so+0x94
#24  0x7f14b5d609e1 in void Gui::_cmdDocument<char const (&) [12]>(Gui::Command::DoCmd_Type, App::Document const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const (&) [12]) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADGui.so+0x3a1
#25  0x7f143b34f2dd in PartDesignGui::TaskDlgFeatureParameters::accept() from /tmp/.mount_FreeCAlilPPK/usr/lib/PartDesignGui.so+0x13d
#26  0x7f143b378010 in PartDesignGui::TaskDlgDressUpParameters::accept() from /tmp/.mount_FreeCAlilPPK/usr/lib/PartDesignGui.so+0x460
#27  0x7f14b6113bc7 in Gui::TaskView::TaskView::accept() from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libFreeCADGui.so+0x67
#28  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f14b30eb6f2]
#29  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libQt5Widgets.so.5(+0x2fab48) [0x7f14b40fab48]
#30  /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7f14b30eb964]
#31  0x7f14b405b522 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAlilPPK/usr/bin/../lib/libQt5Widgets.so.5+0x32
[... cut ...]
#58  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f14b2c46305]
#59  /tmp/.mount_FreeCAlilPPK/usr/bin/freecad(+0x7c2a) [0x55f84da92c2a]
```

### Full version info:

```
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

```

tarman3, 2024-11-30:
Confirm crash FreeCAD.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39315 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

PS
Crash this is of course a problem. But Fillet003 is not the last operation.
In most cases you delete all fillets in the end and create it again.

Most of the added fillets are completely useless and only affect the image on the screen. Over time, I stop adding such operations.

FEA-eng, 2025-10-16:
@leoheck Can you check if you still have the crash? When I open your file on Windows, Chamfer004 looks correct and doesn't break the model or crash FreeCAD.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```