FreeCAD/FreeCAD #16329: [Feature Request]: Create a FreeViewer (and converter)

Mannshoch, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD support a huge amount of filetypes. Way more people need a File Viewer and converter than a CAD.

# Benefit:
- A File Viewer may could also help getting a foothold into companies
- Currently, there is no OpenSource 3D file viewer
- A file Viewer could also be the starting place for new filetype support. Viewing is more easy than edit.
- A viewer need a new clean surface. That is a chance and could be the start of a place for experimenting with surfaces on low complexity and maybe the start for a complete overhaul of the FreeCAD GUI
- For download shown as a **FreeViewer** and after Install it appears as viewer ... but in fact a user may detect that beside **freeviewer.exe** a **freecad.exe** could be found in the application directory. If they open that a full freeCAD appear.  😱
(I do not hope on curiosity but on rumors by others)

# Idea:
https://forum.freecad.org/viewtopic.php?t=90003

Please create a FreeCAD-Viewer (separate download, deployable).
After Installing this variant:

- FreeCAD-Viewer is registered with all currently supported file-types
- Open a supported file-type, do open a window with only a 3D-Area where you see the content
- Open the Application shows a empty window with a Open Button.
- A Menu Button contains:
  - a save button, to save as an other file-type
  - a Send to FreeCAD button that open the file in the full FreeCAD environment.
  - a show property button.

To create this viewer I propose to add this view to the normal FreeCAD Core and decide how to open by the opening command.

Priority list:

1. MVP
   - [ ] Main way for open is double-click a file
   - [ ] Pure 3D view - turn around and zoom
   - [ ] Change view, fit to model
   - [ ] Register file-type on installation.
   - [ ] Deployable installation
     - MSI or installation instruction (without Intune possible)
     - Low Priority `settings=\\share\FreeView.conf` to preset viewer settings.
   - [ ] Wiki how-to insert this app as viewer into other application. (PDM / PLM / DMS / etc.)
1. Windows Explorer integration
   - Preview pane
   - Create an Icon with the 3D-Model as icon.
   - Allow change settings (file registration and Explorer integration)
1. file property dialog - (meta-data per file-type specific -> low priority but prepared to show them)
1. add settings to add an open-with rule per supported file-type (deactivated if changed by deploying)
1. If app started without double-click a file, an empty window with FreeCAD logo as background (may looks like the FreeCAD splash-screen) and an "Open" button appear - drag-and-drop possible.
1. FEM Clipping Plane - (Low Priority: Export dxf from clipped position)
1. Measure
1. Save as ... - convert to file-type

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) Snap 1127
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mno13, 2024-09-06:
"Currently, there is no OpenSource 3D file viewer" -> https://github.com/fougue/mayo

furgo16, 2024-09-06:
Looks great! It would be awesome if they could add FCStd support: https://github.com/fougue/mayo/issues/280

Mannshoch, 2024-09-06:
> "Currently, there is no OpenSource 3D file viewer" -> https://github.com/fougue/mayo

OK thats really cool. 
I would vote to: Merge both project or copy their concept for the viewer or at least do it anyway because of the other Benefits.

maxwxyz, 2024-09-07:
Not sure if this is the right place in this repository. Could you open a forum discussion?