FreeCAD/FreeCAD #16330: Designs created with parts-option look strange

reinaldk, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When combining parts in Freecad these models look different than in the software. In Freecad the parts look as designed, when exporting them to UliMaker or Anycubic slicer part are separated.

Thanks
Reinald

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Austria (de_AT)
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
Without a file there is nothing we can do with this issue. Have you fused the parts into a single solid?

luzpaz, 2024-09-06:
@reinaldk we recommend using the forum to get assistance. Also you are using v0.21.x, and we are on the threshold of releasing v1.0 The issue you may be encountering are fixed. Closing for now.

closed 2024-09-06 by luzpaz