FreeCAD/FreeCAD #16331: BIM: beam bounding box does not correspond to profile size upon creation

furgo16, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD; switch to BIM workbench
2. Set the working plane to top (W, P, 2)
3. Choose the Beam command (B, M)
4. Under `Structure options`, choose `Category: Eurocode timber` and `Preset: 22x100`
5. Click on a point on the 3D window to start the beam
6. Notice that the bounding box's width and length do not correspond to 22mm and 100mm as in the Structure options input boxes. The width of the bounding box seems to match, but the length is wider
7. Click on another point in the 3D window to finish beam creation
8. The beam is created as expected, but differs from the bounding box that was shown on creation

If I then start the command again and choose another preset (e.g. 100x150), the sizes increase as expected, but then length and width are switched. At some point, I believe (because I cannot reproduce it now) that the bounding box did match the 22x100 preset, but unfortunately I cannot manage to reproduce it.

It would seem that the calculation of what the bounding box will look like upon beam creation is somehow out of sync with the beam profile preset.

[image]
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) Snap 1127
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.9.1
  * Curves 0.6.46
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gianlucafiore, 2024-09-08:
The same thing happens to me, and the parameters to be entered in the task panel do not match the measurements of the beam at all. I think it is because it comes from the column plugin, and for example the length of the column is taken as the width of the beam.

furgo16, 2024-09-09:
Good catch. There is something going on where the input for the BIM_Column command (from the toolbar, not the "Column" option on the ratio button) interferes with that of the BIM_Beam.

Watch the drawing of a beam after having created a column:

https://github.com/user-attachments/assets/919ac135-79ae-4c89-aa75-fad68b033cd6

Notice how the height value does not make sense while the beam is being created, yet it's correct after creation.

yorikvanhavre, 2024-09-10:
The combo system that does the same for beams and columns tries to "guess" if we're doing a beam or a column and tries to adapt by switching length and height, but it's always problematic.

It's in my roadmap to refactor that tool to remove that problem by separating both modes more adequately.

But the wrong boundbox seems to be a different issue?

furgo16, 2024-09-10:
> But the wrong boundbox seems to be a different issue?

I do not know enough about the code, but that would be my guess. I cannot tell if the combo system affects it, though.

In any case, from a cursory test, the bounding box for [BIM_Column](https://wiki.freecad.org/BIM_Column) seems to be sized as expected. Whereas the bounding box for [BIM_Beam](https://wiki.freecad.org/BIM_Beam) is not sized as expected, as per the original issue description.

yorikvanhavre, 2024-09-10:
Aah okay... The problem is not the bound box but the preview "box" that appears when drawing the beam.
Indeed I see the problem too, it's not set correctly in Beam mode. I'll have a look at it

furgo16, 2024-09-10:
Sorry for the confusion. I've updated the description to explicitly say preview bounding box.

Roy-043, 2025-09-19:
Related: #15704.